plutoplus/plutoplus #2: Releases & Update to v0.33

Muehsal, 2021-05-31:
Hey there,

according to the readme, releases should be available on this repro. Would you mind adding them for not so experienced users like me?

Second, are there any plans for an update to 0.33?

Best Samuel

SergeyM1969, 2021-06-15:
Hi Samuel !
I am also looking for new FW based on v 0.33 what pushed for official Pluto. However due to the here missing 0.32 release i think Howard not supporting any more this HW, pity. Only the way is burn new FW b own self.
I have tested 0.33 with Pluto+ however all functionality like ethernet etc are lost.

Regards! 
Sergey

howard0su, 2021-06-17:
The code in this repo is 0.33. Are you requesting a binary release? I can release it but it may requires some additional tests.

plantjes1, 2021-06-17:
A full working firmware would be great... happy to beta test for you.
Also adding the network to the f5oeo fork would be great.
I follow that fork but Evariste seems to have stopped for the summer...
Adding the Network socket  to that fork would be appriciated by a large user crowed... using it for dvb...

regards 
Benno

howard0su, 2021-06-17:
What's interesting feature in f4oeo fork? I can help to merge.

plantjes1, 2021-06-17:
Hi Thanks for fast answer..

Pluto is used For QO100 Communication.. Part is Narrow band and sdrconsole.. part is wide band DVB-S2 (TV) For the last parts F5OEO has succesfully adapted the Pluto with DVB firmware making it a great tool foor Transmitting DVB-S  more info on https://forum.amsat-dl.org/index.php?thread/93-adalm-pluto/
Now the Pluto+ solves all problems with standard pluto a,b, c and rev d...
But the Network socket is not supported in the normal Pluto and thus not in the F5OEO with is forked of the V0.32 and also V0.33

Thus the ideal option is adding the gigabyte socket to the F5OEO.. http://firmware.hackhamradio.com/

If you could helpout, i know a large crowed would be happy..
I did some prevew on Youtube compairing the both.. https://www.youtube.com/watch?v=y4b_pCFazK0&t=9s

Thanks sofar

Benno

howard0su, 2021-06-17:
Is this right repo? https://github.com/F5OEO/datvplutofrm

I checked, it contains the change to support network which is not really needed for Pluto+. In additional to that, there are many additional packages pulled into the firmware. I can add those as well. However I have no idea to how to use the firmware. and I will publish one binary for you to play with.

howard0su, 2021-06-18:
Can you try this firmware:
https://github.com/plutoplus/plutoplus/releases/tag/v1.33

plantjes1, 2021-06-18:
Ok, I can confirm this binairy works and is 0.33 Dirty...
It works with sdr console on 145mhz and 2400mhz..
Network ok..

But no additional dvb files jet...
I can use the firmware but have no knolledge of compiling...

I have both pluto with dvb and pluto+ here inhouse so i can do the compaire ..
Now i need to know what you need...

Best regards..
Benno

howard0su, 2021-06-18:
Where can I find the instructions to use DVB? I can check it out to figure
out the missing parts.

howard0su, 2021-06-19:
I believe I didn't start the python web server. let me check what I missed.

plantjes1, 2021-06-19:
Thanks for your support.
More info can be found on this french site..
https://www.f5uii.net/en/patch-plutodvb/

SergeyM1969, 2021-06-19:
Wow Howard, sounds good, 0.33 based FW works fine for me
could you look the link recommended by Benno
https://www.f5uii.net/en/patch-plutodvb/ 
yes probably server not started

SergeyM1969, 2021-06-19:
Howard, sorry, just recognized  in updated FW - no transmission signal in case selected Bandwidth 550khz (SDR Console ver 3.0.27)

howard0su, 2021-06-20:
I will check. For DATV related software, I realized there are several binary files in the repo. I need to reach out to F5OEO to find out if they are opensourced.

plantjes1, 2021-06-20:
Hi, I got little reaction from Evariste. You may need to contact his friend Cristian via https://www.f5uii.net/en/patch-plutodvb/

ploeffler, 2021-08-17:
@howard0su thank you for your work, I am looking forward to get my plutoPlus working for DATV

is there any help I could give you to get that running?
(running FW 0.33 with SDRangel no problems so far)