plutoplus/plutoplus #3: TX always ON

SergeyM1969, 2021-06-12:
I  recognised that TX always enabled , on PS_MI00 are always 1,8v  , probably FW fault or?
pls assist

Sergey

plantjes1, 2021-06-17:
Same here.. PTT not supported on the FW as it seems...

howard0su, 2021-06-17:
How is the original FW from ADI behaving? I don't think I did anything different for that I/O pin.

plantjes1, 2021-06-19:
Was it implemented in ADI?

SergeyM1969, 2021-06-19:
Howard, honestly speaking no idea in case MI00 signal supporting in original FW
what are gpio specification for ADI FW are  exist ? 
In case Pluto+ have such signal out i think should be gpio defined

howard0su, 2021-06-20:
I need to dig out more. in original firmware, it is also defined as GPIO in FPGA code. However I don't know about the software part in Linux.

do6ep, 2022-09-19:
Same here... 
PS_MI00 and PS_MI11 are always at 3.3V. I tried every firmware I could find.
I tried also toggling the GPOs from the console in manual mode but they just won't toggle to 0V.
Is this normal behavior?
What's the magic to make the PTT toggle?

Is there maybe an other way to get an PTT out of Pluto+ ?

plantjes1, 2022-09-19:
Same here, No activity on the PTT. 
I  have an old PTT way working on my first pluto witch worked on pre F5OEO versions...
It involved a data line between processor and TX  controll whitch is detected with a diode.

It may be whise to check that option as the IO seems bussy.

Benno

do6ep, 2022-09-19:
I have also found this alternative method, but until now I could not find out where exactly this data line can be found on PlutoPlus.

-->> https://wetzsteinfunker.de/wordpress/ptt-output-fuer-den-adalm-pluto/

But it would be nicer to be able to use the already built-in PTT circuit, after all that was one of the reasons for my purchase.

______________________________________________________________________

Auch ich habe diese alternative Methode schon gefunden, aber bis jetzt konnte ich leider nicht herausfinden, wo genau am PlutoPlus diese Dataline zu finden ist.

-->> https://wetzsteinfunker.de/wordpress/ptt-ausgang-fuer-den-adalm-pluto/

Aber schöner wäre es jedoch, die bereits intern verbaute PTT-Schaltung nutzen zu können, immerhin war das für mich einer der Kaufgründe.

do6ep, 2022-09-22:
I was now able to find the required TX_D0 line on the PlutoPlus (on the back of the board, under the processor) and carry out the modification from https://wetzsteinfunker.de/wordpress/ptt-OUT-fuer-den-adalm-pluto/.
However, the switching of the PTT is also delayed by about 500-800ms, which I don't like.
_______________________________________

Ich konnte nun die benötigte TX_D0 Leitung am PlutoPlus (auf der Rückseite der Platine, unter dem Prozessor) finden und die Modifikation von https://wetzsteinfunker.de/wordpress/ptt-ausgang-fuer-den-adalm-pluto/ durchführen.
Jedoch ist das Schalten der PTT hierüber auch ca. 500-800ms verzögert, was mir nicht gefällt.

[image]
[image]
[image]

plantjes1, 2022-09-22:
Whooo, Great job.,
I understand the delay.
I think its normal as there is always a delay in TX button on pc and processed TX. 
One experience on my old pluto's in compairisment with the F5UII way (where 2 data controll lines been used) is some switching when the pluto boots. and that is due to the callibration the firmware does at booting.
This callibration makes irratic signals with have blown the preamp stages in the past. So make a hard controll switch to prevent unwanted PTT switchings...

Thanks for your contribution, I will likely follow these steps...
Best regards
Benno

mx0xuu, 2022-09-28:
Recently purchased Pluto+ and has the same issue with PTT TX always on. 
Is there a fix for this otherthan the hw mode suggested above ? Are there any updated fw > 0.33 ?

Thanks,
Gopan

do6ep, 2023-01-16:
Hello, is there any progress in the meantime?
Why equip the Pluto+ with an extra PTT circuit if it doesn't work?

jgreusel1, 2023-04-27:
I would like drive a rx/tx relay so I can use the same antenna for both functions but from what I'm reading here this may not work with current PPT/firmware arrangement. Has anyone sorted this out?

jgreusel1, 2023-04-27:
Sdr Console (Simon Brown) offers this solution from software which seems straight forward so I may try this switching method.
https://www.sdr-radio.com/relays

KC9OJV