plutoplus/plutoplus #4: running gnu radio on pluto plus?

TheRealMoeder, 2021-06-18:
Hi Howard,

long time no see. Since you star'd F1ATB/RemoteSDR repo I noticed your recent SDR involvements. I think F1ATB has started something nice, but I was wondering if we could run the webserver and gnu radio flows inside the Pluto(plus) instead of utilizing yet another SBC. What do you think?

howard0su, 2021-06-19:
Nice to see you. Surprise to see we are again working on a different hobby project.

I think so. technically it is not hard. Since Pluto+ is using buildroot, you can leverage it to compile gnuradio there. Webserver, you have lots of choices even python which I already compiled it into. Google "gnuradio buildroot" for more informaiton. And check patches/buildroot.diff to find out how we add the additional packages.

TheRealMoeder, 2021-06-19:
I will try to look into it. I'm wondering if the Zynq-7000 is capable enough to handle everything. I only own the original Pluto, not your improved version, but from what I see software differs only marginal, so it could work in either device.

howard0su, 2021-06-20:
I think so. The Pluto+ adds the ethernet and sd card and replaced with a different footprint of 7010. From software perspective, there is not much difference.

eljuligallego, 2021-11-03:
Hi, here you can find the gnuradio inside the adalm pluto:

https://ez.analog.com/adieducation/university-program/f/discussions/98761/gnu-radio-on-the-plutosdr-proof-of-concept/296003#296003

the experience is not very good, the cpu is very slow for the GUI...