plutoplus/plutoplus #5: Question not issue - 'portapack'...

ehagerty, 2021-06-24:
Hi, this is an awesome design, I wish I had your skills!  Forgive the direct question, but do you think this design might be able to support a touchscreen or similar a la the 'portapack' concept possible with the hackrf1?

Thanks so much for your time.

waterwin, 2021-07-08:
See this : https://github.com/g4eml/Langstone

ehagerty, 2021-07-09:
Thank so much!  I don’t suppose you will be making .sch and .bed files available at some time?  I’m keen to see if I can design something like a portapack for this board specifically. 😊

waterwin, 2021-07-10:
You will have to ask the developer, I only pointed you there as it might answer your questions. I have not used this material myself.
Kind regardsErwin