plutoplus/plutoplus #6: Pluto+ starts up only when first press and hold DFU-button few seconds (Jumper: URST/MIO46)

dc1saf, 2021-08-03:
Hey, everyone,

thanks to everyone who made this great piece of hardware possible :-)

Unfortunately I run into very strange issue: When I plug in the USB-cable, there ist just the green LED  going on. Nothing more will happen. Especially no drive will appear and I am not able to establish a connection via SDR Console or PuTTY. Jumpersetting: URST/MIO46.

I only get the Pluto+ running, when I press the DFU button for some seconds, while plugging in the USB cable. When the green LED goes off, wait a short moment and releasing the DFU-button. Then there is a chance the Pluto+ will run normally: it appears as a drive, it is accessible with SDR Console and PuTTY.

If you select URST/MIO52, you might enter the DFU-Mode without any problems, but if you do not press the DFU button, the Pluto+ will not start up.

What did I do just right before this issue appears:
I was trying to assign a different IP address in the [NETWORK] section in config.txt, so I could connect it into my network setup. It was accessible via LAN afterwards with the assigned static IP, then one time it got a DHCP IP assigned from the router and finally, the default ip adress 192.168.2.1 was again in config.txt - I don't know why and how it was changed back.

What did I try to regain "normal" behaviour:
* reinstalling Firmware via DFU (with UPDATE.exe, Windows 10) (Jumper: URST/MIO52)
(at the moment I have installed the 20 MB v0.33 dirty)
* open config.txt, save the file and eject the PlutoSDR drive

 I have no idea how to get it again to start up normally without playing with the DFU button. Does anyone have any ideas?

Best ragards,
  Fabian

fei666888, 2021-09-02:
Have you tried to update the firmware? Normally, if you don’t need to press DFU to start, it should start automatically after power-on

dc1saf, 2021-09-02:
The upcoming days I tried it several times and with various images. After doing so for days, it finally woke up again :-)
I don't know what has happened, whether why I got it to be bricked, or why I had to upload the firmware several times.

Normally, I really dig deep into why something has happened for learning purposes, but this time, I have completely no clue whatsoever. Nevertheless: It's working again :-)

Thank you fei666888 for your reply.

closed 2021-09-02 by dc1saf