plutoplus/plutoplus #7: Undefined usb_ethernet_mode definition bricks device

K7MDL2, 2021-08-10:
Received a new PlutoPlus today and tried to put it on the wired network.  2 items in this report.  1st is the problem resulting in a completely unresponsive device, the 2nd part is the attempts to recover that did not work.

I was trying to change the ethernet address since 192.168.2.1 is my network's router.  Logged in via terminal, used fw_setenv commands top change the ipaddr.  That was not working, was briefly getting random IP addr values as logged by my router, they were never around long enough to get a ping.  Tried fw_setenv usb_ethernet_mode rndis on a guess.  No change.  I had read in one article that leaving it blank was the same as rndis setting, so tried that.  The commend line never returned.  fw_printenv commands verified each change I made with fw_setenv.

At that point reboots resulted in USB device failed to reset and failed to enumerate errors.  LED solid bight green never changes.  No mass storage, no response.  Tried different USB ports (USB3 power hubs, direct on computer, USB2 hub), tried changing jumper to MIO52 and holding DFU during power up, no responses, not even bad device USB error messages as before.  At this point after many attempts, it no longer even gives me unknown device errors, completely unresponsive, no network presence either.

Also created a config.txt on a SD card.  All I had was a 128GB.  Formatted it exFat.  Also placed the 0.33 DFU files on it hoping it might see those.  Nothing.

I have a JTAG programmer on order so will be a few days before I can try that recovery path.  Given that things quickly deteriorated, I am not expecting it to be accessible.  

Also have 16GB micro SD cards coming in case 128is too large.  I could never find a spec on the supported card sizes.  That should be added to the the spec listings.

MatthiasDirlinger, 2021-08-24:
Have the same Problem could you recover the Device via JTAG?

K7MDL2, 2021-08-24:
I have not been able to build any images to flash, one error after another, i have abandoned that approach. I found one boot related image and it turned on a blue led, but still get usb device failed to reset error. Also the JTAG uart terminal never shows any signs of life. Working on sending it back for repair.

Heertsch, 2021-08-27:
same here: After changing the ip-address via config.txt all ports are blocked. (This behavior is the same as  with the original Pluto.) Then I installed Adalm-fw 0.34 via copying it to the Pluto+ drive. Does not help.
But after pressing DFU (without changing the jumper) Pluto+ is invisible for USB and the ethernet port responds no ping. So I can't access Pluto+ 
Any workaround?

K7MDL2, 2021-08-27:
I later hooked up the ADALM-JTAG programmer with UART and there is no activity on the UART,  I was never able to get the Xilinx build to work so gave up on that.  Has the symptoms of hardware failure with zero boot activity and no USB, no DFU action.  Had to give up on this one.  I hate losing to a machine. :-).

MatthiasDirlinger, 2021-08-28:
Could it be some Change of Hardware? my Pluto+ has 3.3V instead of 1.8V on the JTAG connector and some IOs have 3.3V and some have 1.8V, DFU Button has Level 3.3V is there a chance that Jumper in Wrong Position + DFU Kills the Hardware?

Heertsch, 2021-08-30:
>  is there a chance that Jumper in Wrong Position + DFU Kills the Hardware?
That sounds plausible! So a rule should be: 
### NEVER HIT THE DFU-BUTTON 
# if the jumper is in the wrong position.
And to the manufakturier: Change your layout

MatthiasDirlinger, 2021-08-30:
and the wrong position is?
i think i have maybe pressed dfu and the jumper was on MIO46 and URST when tryed DFU
does this kill the hardware?
and where is the failure in layout?

MatthiasDirlinger, 2021-08-30:
Have my Pluto+ running again
its no Hardware fault
used Xilinx Vivado to programm Bitstream
transfered u-boot.elf from original Pluto Repository
after dfu-util transfer of boot.dfu, uboot-env.dfu and pluto.dfu its up again

MatthiasDirlinger, 2021-08-31:
I think there is a new Hardware release incompatible with this Firmware
Run Pluto Original Firmware on device without any Problems
when flashing Plutoplus firmware from here uboot is killed
both variants tryed with DFU and over USB Storage
the difference i see is JTAG has 3.3V instead of 1.8V and the Memory Chip is a W25Q256
I can unbrick the Pluto+ with JTAG without any problems running orginal Firmware but no chance to get Pluto+ Firmware running

Heertsch, 2021-08-31:
So there is a chance to unbrick with JTAG? 
K7MDL2:
> I later hooked up the ADALM-JTAG programmer with UART and there is no activity on the UART, I was never able to get the Xilinx build to work so gave up on that. 

Did you try to use the SD-card slot? (I have no JTAG til now.)

K7MDL2, 2021-08-31:
I did not have a bootable SD card. Mine did not come up enough to initialize USB, so no mass storage, no DFU boot, no nothing.

MatthiasDirlinger, 2021-09-01:
JTAG works for me made an Adapter for it [JTAG ADAPTER](http://tiger.dirlinger.at/plutoplus)
SDcard does nothing because the Bootloader is destroyed
You need Vivado SDK an older Version with XSCT and plutosdr-jtag-bootstrap files
in XSCT you can upload the bootstrap uboot with JTAG

Jumper placed on JTAG

connect
target 2
source ps7_init.tcl
ps7_init
ps7_post_config
dow u-boot.elf
con

then the device should boot up with uboot in RAM

Jumper removed on JTAG

after this press the DFU button and upload the Firmware with normal DFU
i made the failure the first time leave the Jumper on JTAG then the DFU dont programms into Flash

Now you should have a running "Adalm Pluto" you can change firmware over DFU and Storage in all variants from Original Firmware
But when i try to install PlutoPlus Firmware Bootloader is killed again
I think there is some Problem in the PlutoPlus firmware with flashing

fei666888, 2021-09-01:
If DFU cannot be started normally (confirmed that URST has been connected to MI052 via jumper), the boot part may be damaged, you can try JTAG or directly reprogram the part of boot.bin of QSPI (W25Q256) through the programmer
[boot.zip](https://github.com/plutoplus/plutoplus/files/7088903/boot.zip)
[PLUTOX_SDR-V2.0-20210707.pdf](https://github.com/plutoplus/plutoplus/files/7088906/PLUTOX_SDR-V2.0-20210707.pdf)

MatthiasDirlinger, 2021-09-01:
do you have the boot.bin file as bootstrap for jtag?

fei666888, 2021-09-01:
It seems that JTAG cannot write that boot partition

MatthiasDirlinger, 2021-09-01:
how to programm the chip without unsoldering?
fine would be a manual how to programm the complete device
i can write the original uboot from adalm pluto with jtag but need the bootstrap format

fei666888, 2021-09-01:
Once the DFU device is successfully identified, pluto. DFU and uboot-env.dfu can be rebrushed with the update. script
[plutosdr-fw-v0.31.zip](https://github.com/plutoplus/plutoplus/files/7089033/plutosdr-fw-v0.31.zip)

MatthiasDirlinger, 2021-09-01:
thanks your firmware does the trick to load 0.32
but if i then update to 0.33 from here the bootloader is destroyed again
the question is why is the firmware from here the wrong for the V2 hardware release?

fei666888, 2021-09-01:
Let's confirm later

Heertsch, 2021-09-01:
MatthiasDirlinger:
> JTAG works for me made an Adapter for it JTAG ADAPTER

great! Do you have another one for me? => https://hb9hci.support/Pages_DE/kontakt.html

fei666888, 2021-09-01:
We have reproduced this issue, after flashing the 0.33 version of the firmware, there will be a problem of not being able to start. Everything is normal on the 0.32 version. We strongly recommend not to update the 0.33 version of the firmware.
https://drive.google.com/file/d/10zrF4lcKLt0HbGna57fAqCa6EFLA0IsY/view?usp=sharing

K7MDL2, 2021-09-01:
Can you test what happens when you set the parameter usb_ethernet_mode= <no value>?  Doing this caused the terminal command line to stop working and bootup to fail.

https://drive.google.com/file/d/10zrF4lcKLt0HbGna57fAqCa6EFLA0IsY/view?usp=sharing<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdrive.google.com%2Ffile%2Fd%2F10zrF4lcKLt0HbGna57fAqCa6EFLA0IsY%2Fview%3Fusp%3Dsharing&data=04%7C01%7C%7C70bbbfa869bc4d30243608d96d645a9d%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637661100047006035%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=kll36khSHAY0NfuOG4rEU9KCUJRgaGQlGaSm3%2Fkh1CI%3D&reserved=0>

MatthiasDirlinger, 2021-09-02:
Have tryed first with VIM and then with Windows Notepad
after reboot the Option is removed out of the config File and nothing bad happens
I think this Issue can be closed and we should open an Issue 0.33 Firmware is bad and destroys Hardware

matthart, 2021-09-02:
My new Pluto+ even did show up on my PC, when trying to start it for the first time. The green LED is permanently on while trying this. That is why I am expecting that the latest Pluto+ firmware is already installed on it. Did you succeed, when trying to get your Pluto+ working, again? And if so, how did you manage to do this? 

> Can you test what happens when you set the parameter usb_ethernet_mode= <no value>? Doing this caused the terminal command line to stop working and bootup to fail. Sent from my T-Mobile 4G LTE Device Get Outlook for Android<https://aka.ms/ghei36>
> […](#)

MatthiasDirlinger, 2021-09-02:
I postet manual some Comments above
you need xilinx sdk 2019.1
then a jtag-hs3 and a adapter board or crasy wires :)
then the commands and jumper changing after this it should be alive again i will upload the needed files for full recovery to the folder i have postet

MatthiasDirlinger, 2021-09-02:
I cannot reproduce to kill the Device with config.txt have wild messed the File and its Booting up with restored file

K7MDL2, 2021-09-02:
To be clear, the original problem described in this Issue was the terminal session becoming unresponsive after giving the terminal command

"fw_setenv usb_ethernet_mode="

This is via the terminal, not via editing the config.txt file.

After that my device never again booted up, USB was never initialized (Failed to reset among other errors seen) and the green light stays on bright.  No jumper changes, no JTAG uart terminal activity.  I did not find a suitable set of JTAG files to load up, was not able to successfully build them from source in may attempts.  Given the symptoms, the USB functionality seemed to not work.  I have since returned my device so cannot try other experiments.

I had successfully used the "fw_setenv usb_ethernet_mode=RNDIS" and other valid values OK, it was when I left the value blank the trouble started.  Likely a lack of validation in the fw_setenv command parsing.

Good find on the 0.32/33 issue.

Can you test what happens when you set the parameter usb_ethernet_mode= ? Doing this caused the terminal command line to stop working and bootup to fail. Sent from my T-Mobile 4G LTE Device Get Outlook for Androidhttps://aka.ms/ghei36<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Faka.ms%2Fghei36&data=04%7C01%7C%7C0e787c6264364909ca4a08d96ddc802e%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637661616067781164%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=a%2BRA3Dtef0clvm3wEGf052kjn9plRFwpwVM8kmEDIkA%3D&reserved=0>
...

-

MatthiasDirlinger, 2021-09-02:
fw_setenv usb_ethernet_mode= does nothing as it unsets the variable
fw_setenv usb_ethernet_mode = sets usb_ethernet_mode to value = -> Device boots
fw_setenv usb_ethernet_mode =foobar sets usb_ethernet_mode to value =foobar -> Device boots
fw_setenv usb_ethernet_mode foobar sets usb_ethernet_mode to value foobar -> Device boots

Find no way to brick the Device with these Settings

fw_setenv are not with = instead you should use a blank
fw_setenv usb_ethernet_mode RNDIS is the correct command

K7MDL2, 2021-09-02:
Possible my particular unit failed at that particular time and since it was the last command given, took the blame 😊

K7MDL2, 2022-03-26:
I tried to get a non-working PlutoPlus unit running over the last 2 days and have reached partial victory.  It is now working on PlutoPlus jumper mode (URST-MIO46) with ethernet over USB only with SDRSharp and SDRConsoleV3.  

Many hours of brute force trial and error controlling each variable.  My first step was to get the legacy mode working and DFU utility working which I eventual did.  Cannot tell you exactly how or why though it was very hit or miss even making no changes.  It either booted up or did not.  Not knowing yet what I have documented below I think part of the issue was the sheer number of USB "ejects" and reboots and power cycles combined with the occasional USB mass storage folder eventually appearing and then editing the config.txt to suggested values then trying other values until something worked.

Loading up firmware 0.32 and 0.34 had mixed results, mostly related to the config.txt ethernet setting and whether I had the ethernet cable plugged in or not.  Neither ever worked in Plus mode until I loaded the 0.34 version mentioned in [Issue #14.](https://github.com/plutoplus/plutoplus/issues/14) The direct link to the firmware is http://tiger.dirlinger.at/plutoplus/34plus/ 

Also I found I cannot set the host IP address to the actual valid host ip address.  I had to set it to something not used by anything,

So here is my list of knowledge so far to get my unit here working in Plus mode.

1. Cannot have USB cable and ethernet cable plugged in at same time with both configured.  
2. It seems to always use ethernet over USB if you have the USB cable in the data side connector ignoring the hardware enet connection.
3. Cannot set the ip_addr_host address to a valid Host PC address as it is supposed to be.
4. I have not seen the ethernet jack alone work yet, only success is over USB virtual ethernet connection.
5. I have never seen ping work from the serial console in the unit to any address outside of the box.
6. If USB_ethernet is blank it will do DHCP.  I had a DHCP reservation.
7. If you cannot ping it, it won’t work for RF.
8. I found the firmware mentioned in Recovery ADALM PLUTO FW to PLUTO plus FW · Issue #14 · plutoplus/plutoplus (github.com) works. [http://tiger.dirlinger.at/plutoplus/34plus/.](http://tiger.dirlinger.at/plutoplus/34plus/.)  The 0.34 version from this github site did not work in plutoplus mode for me, only legacy mode.  No idea what is different.
9. To get the full range you need to set the compatibility to AD9364 or AD9361 per normal instructions. [Customizing the Pluto configuration [Analog Devices Wiki]](https://wiki.analog.com/university/tools/pluto/users/customizing?msclkid=46708f35aced11ec814696c4ef57c34f)
10. DFU button must only be pressed when jumper is on legacy mode position, URST-MIO52 pins.  Can use standard DFU-Uitil tooling to upload new firmware.
11. Can use UDisk to easily upload new firmware instead of the DFU utility.  Just drop the boot.frm and pluto.frm files onto the root of the USB mass storage folder and do a proper USB device eject.

I want to make the ethernet jack work properly without an USB connection so still have more work to do there.

K7MDL2, 2022-03-29:
Update, after experimenting with manually configuring the ethernet and usb network addresses to put then on separate subnets, and then editing the config.txt to give the same result, I now have the unit working on the ethernet port only with no USB data connection, just using the 5V power port.

K7MDL2, 2022-03-29:
It is worth noting the USB ethernet interface remains configured even though there is no data connection (with IP addresses defined in config.txt).  I think this was part of the confusion as the real enet port will drop to auto-ip on the 169 subnet and not interfere with the USB network.  I had both ports on the same subnet assuming the unconnected one would not be active.  The USB is always active and if not connected acts as a bit bucket.  I have to go back and try a blank ethernet over usb setting ana see what happens.