plutoplus/plutoplus #10: !!! Dont Install 0.33 on Pluto+ board Release V2 it bricks the Device

MatthiasDirlinger, 2021-09-02:
A Pluto+ Release V2 has 3.3V on JTAG connector instead of 1.8V
If you upload 0.33 on these devices the Bootloader is destroyed and you have a Bricked device
**!!! Be Warned Recovery is not so easy !!!**

Here are the needed Files [http://tiger.dirlinger.at/plutoplus/](http://tiger.dirlinger.at/plutoplus/)

robaol, 2021-09-05:
Is this the same fw version referred to in [issue 2](https://github.com/plutoplus/plutoplus/issues/2)? In that issue v0.33 is reported to be working...?
(I'm not disputing the advice, not to load v0.33, I'm just confused about the different reports.)
Also is the 3.3V selection on the JTAG software selectable? I assumed it was hardwired in the board design?
I am considering whether to get  a pluto+...

MatthiasDirlinger, 2021-09-05:
If the connector has 3V3 TMS TCK TDO TDI GND its Revision 2
if the connector has 1V8 TMS TCK TDO TDI GND its Revision 1

tiborkiss, 2021-09-19:
Here is a Revision 2 board.
[image]

If I cannot upload 0.33, then is it possible to upload 0.34 directly? Or just by the files you (@MatthiasDirlinger) provided?
Are there in your http://tiger.dirlinger.at/plutoplus/34plus/ any patches relative to 0.34 official?
Have you had to do anything special to upload 0.34?

_I just got one and my intention is mainly to use as a RF protocol debug tool, probably I will dig into the programming of the firmware also. (at least +1 people who is trying to keep up-to-date this board's fw)._

MatthiasDirlinger, 2021-09-19:
The firmware in this folder is original firmware patched with network
It is running i tryed it some days but cannot test at the moment because a dcdc converter on my board died
Iam looking for replacement at the moment
I hope there is coming some official support from here i have not planned to develop firmware

mbehn1976, 2021-09-26:
[image]
From the manual to recover - is the jumper from JTAG to GND the correct one?
[image]
And from your point of view should this XILINX USB - JTAG device work?

tiborkiss, 2021-09-27:
Thank you Matthias for the details.
One thing I still did not understood completely. Only 0.33 is the dangerous firmware, but 0.34 already works? I mean, if we just skip 0.33 and directly 0.34 we get anything solved?

mbehn1976, 2021-09-27:
Matthias, I tried your approach and xsct seems to work fine:
xsct% connect                                                                   
tcfchan#0
xsct% target 2                                                                  
xsct% source ps7_init.tcl                                                       
xsct% ps7_init                                                                  
xsct% ps7_post_config                                                           
xsct% dow u-boot.elf                                                            
Downloading Program -- /home/mario/Documents/pluto-plus-sdr/tiger.dirlinger.at-plutoplus-34plus/from-website/u-boot.elf
	section, .text: 0x04000000 - 0x0403f333
	section, efi_runtime_text: 0x0403f334 - 0x0403f34b
	section, .rodata: 0x0403f350 - 0x0404f37a
	section, .hash: 0x0404f37c - 0x0404f393
	section, .dtb.init.rodata: 0x0404f3a0 - 0x0405175f
	section, .data: 0x04051760 - 0x04053987
	section, .got.plt: 0x04053988 - 0x04053993
	section, efi_runtime_data: 0x04053998 - 0x04053a5f
	section, .u_boot_list: 0x04053a60 - 0x04054863
	section, .rel.dyn: 0x04054864 - 0x0405d2cb
	section, .bss_start: 0x04054864 - 0x04054863
	section, .bss: 0x04054864 - 0x040a52b3
	section, .bss_end: 0x040a52b4 - 0x040a52b3
100%    0MB   0.1MB/s  00:05                                                    
Setting PC to Program Start Address 0x04000000
Successfully downloaded /home/mario/Documents/pluto-plus-sdr/tiger.dirlinger.at-plutoplus-34plus/from-website/u-boot.elf
xsct% Info: ARM Cortex-A9 MPCore #0 (target 2) Stopped at 0xffffff28 (Suspended)
xsct% con                                                                       
Info: ARM Cortex-A9 MPCore #0 (target 2) Running

But the dfu-util does not see it (also not with dfu-util -l or sudo dfu-util -l):
mario@pop-os$ ./dfu-util -D boot.dfu -a boot.dfu 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0456
Match product ID from file: b673
dfu-util: No DFU capable USB device available

Any idea what I am doing wrong or how to fix it?

MatthiasDirlinger, 2021-09-28:
do you have serial console on the board? what it shows?
you have to press and hold the dfu button when sending con and the jumper for dfu has to be in the right position

mbehn1976, 2021-09-29:
Thank you Matthias, pressing and holding the DFU button was the key, with the USRT-MIO52 jumpered for DFU. Then device shows up with dmesg as following:
_[ 2194.445537] usb 5-2: new high-speed USB device number 8 using xhci_hcd
[ 2194.597243] usb 5-2: New USB device found, idVendor=0456, idProduct=b674, bcdDevice= 2.21
[ 2194.597256] usb 5-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2194.597259] usb 5-2: Product: USB download gadget
[ 2194.597263] usb 5-2: Manufacturer: Analog Devices Inc._

A "dfu-util -l" gives:
_dfu-util 0.9
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Device has DFU interface, but has no DFU functional descriptor
Deducing device DFU version from functional descriptor length
dfu-util: Cannot open DFU device 0456:b674_

But with "sudo ./dfu-util -l" you are back in the game:
_dfu-util 0.9
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
Found DFU: [0456:b674] ver=0221, devnum=8, cfg=1, intf=0, path="5-2", alt=4, name="spare.dfu", serial="UNKNOWN"
Found DFU: [0456:b674] ver=0221, devnum=8, cfg=1, intf=0, path="5-2", alt=3, name="uboot-env.dfu", serial="UNKNOWN"
Found DFU: [0456:b674] ver=0221, devnum=8, cfg=1, intf=0, path="5-2", alt=2, name="uboot-extra-env.dfu", serial="UNKNOWN"
Found DFU: [0456:b674] ver=0221, devnum=8, cfg=1, intf=0, path="5-2", alt=1, name="firmware.dfu", serial="UNKNOWN"
Found DFU: [0456:b674] ver=0221, devnum=8, cfg=1, intf=0, path="5-2", alt=0, name="boot.dfu", serial="UNKNOWN"_

mbehn1976, 2021-09-30:
Actually some more questions a. how do I get the serial console attached to the board - via Vivado? b. is the gigabit network port working, so I could stream data faster of the pluto+ instead of being limited with usb 2 bottleneck?

ChromeBumper, 2022-08-28:
It's been almost a year, is this project dead? I have a bricked board and it looks like I need to buy a jtag cable/adapter to revive it.

ChromeBumper, 2022-08-29:
update: I thought my board was bricked because it was not showing up as a mass storage device or on lsusb. It turns out that once I switched the jumper over to the URST-MIO46 position it now works as expected. The second channel is not recognized by the Python API using the scripts that various how-to articles list as a test but both channels do work using the Gnuradio FMComms2/3/4 block.

matthart, 2022-09-07:
> The firmware in this folder is original firmware **patched with network** it can be installed via dfu or drive It is running i tryed it some days but cannot test at the moment because a dcdc converter on my board died Iam looking for replacement at the moment I hope there is coming some official support from here i have not planned to develop firmware

Is there such a version for the latest v0,35 available?

nmaas87, 2022-11-29:
I tried this with @MatthiasDirlinger 0.34plus firmware and got the device reset to that, however I only see an xadc device right now and no ad9361 anymore. Furthermore dmesg tells me it has issues loading the device:
````
ad9361 spi0.0: ad9361_probe : enter (ad9363a)
ad9361 spi0.0: Failed to set BB ref clock rate (-22)
ad9361: probe of spi0.0 failed with error -22
````

any ideas how to fix that? would be very greatful, thank you!

````
Library version: 0.24 (git tag: c4498c2)
Compiled with backends: xml ip usb serial
IIO context created with usb backend.
Backend version: 0.23 (git tag: v0.23)
Backend description string: Linux (none) 5.4.0-00535-g9c04de11ae53-dirty #1 SMP PREEMPT Thu Sep 2 21:07:28 CEST 2021 armv7l
IIO context has 15 attributes:
        hw_model: Analog Devices PlutoSDR Rev.C (Z7010-)
        hw_model_variant: 1
        hw_serial: 10400094c2950017f3ff23001c1c0c64cf
        fw_version: v0.34-dirty
        ad9361-phy,xo_correction:
        ad9361-phy,model:
        local,kernel: 5.4.0-00535-g9c04de11ae53-dirty
        uri: usb:2.11.5
        usb,idVendor: 0456
        usb,idProduct: b673
        usb,release: 2.0
        usb,vendor: Analog Devices Inc.
        usb,product: PlutoSDR (ADALM-PLUTO)
        usb,serial: 10400094c2950017f3ff23001c1c0c64cf
        usb,libusb: 1.0.22.11312
IIO context has 1 devices:
        iio:device0: xadc
                10 channels found:
                        voltage5: vccoddr (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 2044
                                attr  1: scale value: 0.732421875
                        voltage0: vccint (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 1395
                                attr  1: scale value: 0.732421875
                        voltage4: vccpaux (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 2459
                                attr  1: scale value: 0.732421875
                        temp0:  (input)
                        3 channel-specific attributes found:
                                attr  0: offset value: -2219
                                attr  1: raw value: 2619
                                attr  2: scale value: 123.040771484
                        voltage7: vrefn (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 2
                                attr  1: scale value: 0.732421875
                        voltage1: vccaux (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 2464
                                attr  1: scale value: 0.732421875
                        voltage2: vccbram (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 1395
                                attr  1: scale value: 0.732421875
                        voltage3: vccpint (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 1396
                                attr  1: scale value: 0.732421875
                        voltage8:  (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 3690
                                attr  1: scale value: 0.244140625
                        voltage6: vrefp (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 1706
                                attr  1: scale value: 0.732421875
                1 device-specific attributes found:
                                attr  0: sampling_frequency value: 961538
                No trigger on this device
````

PeterBeijer, 2023-05-29:
@nmaas87 did you do a reflash with the JTAG cable etc.? I have the same problem s you have: the AD9361 not showing up.  I wonder is it worth buying the JTAG cable to solve this problem?

regards,
Peter

nmaas87, 2023-05-30:
No, I don't think a JTAG cable will help. I might be wrong, but I think with each update the new Bitstream for the FPGA will be sent to the device and the bitstream/software will be loaded each boot into the FPGA. 

In the end, it looked like an external clock reference from another FPGA could have helped, but I am unsure: https://github.com/plutoplus/plutoplus/issues/26#issuecomment-1567776189

wolfgang6444, 2024-05-27:
I am confused: Apparently I have a pluto+ rev2 (3.3V) on the pinheader. I need to unbrick via JTAG. Can I directly interface with digilent HS2? 
JTAG levels are still 1.8V dispite the 3.3V on the pin?

nmaas87, 2024-05-27:
I do not know if JTAG levels are 1.8v on the revision 2, I do not have any JTAG device - but as said, the FPGA Bitstream is part of the DFU package - so its probably better to just unbrick via DFU.

wolfgang6444, 2024-05-28:
I'd love to use DFU, but that is frozen too. So I guess JTAG is my last resort.
Measuring the PU-levels on the connector, IO voltage for JTAG is 1.8V, while UART is 3.3V.
Having two different IOs with different voltage levels sharing the same pin-Header and then suggestively providing the higher of the two  supply-voltages on a pin of this header is an 'interesting' design-choice to say the least.
I guess I will have to add a 1.8V LDO.
BTW.: Does somebody know if the HS2 interface board can be used, or do I have to get an HS3? According to Digilent, the difference is, that HS2 cannot put the controller into reset.
Many thanks,

Wolfgang