plutoplus/plutoplus #12: New board revision and fw updates??

robaol, 2021-09-30:
[issue #10](https://github.com/plutoplus/plutoplus/issues/10) shows the v2 board has an error in the voltage on a link pin (3v3 instead of 1v8).
@howard0su  do you plan on corecting this? If I buy a board on line at the moment, which board version do I get?
What level of support do you plan for the firmware,  given, for example, that ptt appears not to work, issue X (Sorry I can't remember the number)?

fei666888, 2021-10-09:
The V2 version of the hardware has replaced the 3.3V SPI-FLASH, and the JTAG pin voltage has been changed from 1.8V to 3.3V. The firmware is compatible with the previous one.

fei666888, 2021-10-09:
The ones that are purchased online now, if not surprising, are the v2 version of plutoplus