plutoplus/plutoplus #14: Recovery ADALM PLUTO FW to PLUTO plus FW

SQ6RDP, 2021-11-09:
If You loaded  FW from official site istade of pluto plus FW, ...just chcnge jumpers :
 to this:
**When using the official ADI-PlutoSDR firmware, please connect to URST-MI052.**
When using Pluto+ firmware and need to support Ethernet, please connect URST-MIO46
When using SD card to boot, please connect SD-H to 1V8
When using JTAG to debug, please connect JTAG# to GND
Please connect EXCLK to GND when using external reference clock input

load FW once again by USB port with :
http://tiger.dirlinger.at/plutoplus/34plus/     - pluto.frm

back jumper to:

**When using Pluto+ firmware and need to support Ethernet, please connect URST-MIO46**

Than Your pluto is Back!

DeonMarais64, 2022-10-25:
@SQ6RDP Have you by any chance done a build for 35plus ?

Argonghost, 2024-02-21:
File "/home/argonghost/.local/lib/python3.10/site-packages/adi/rx_tx.py", line 184, in _rx_init_channels
    raise Exception(f"Channel {m} not found")
Exception: Channel voltage2 not found