plutoplus/plutoplus #16: 1PPS İnput

u-erdemir, 2021-11-18:
Hi,
My orders have arrived yesterday. It is a very good upgrade on the original Pluto. In Pluto, there is no pps input to time synchronizate multiple devices. Have you ever tried feeding pps into board and synchronize multiple devices?
Thanks.

fei666888, 2021-12-08:
In theory, any IO connected to the FPGA can be used as the input of the PPS, but this requires attention to the level matching.I think you can synchronize multiple devices by directly using a clock source and inputting through CLK_IN on plutoplus.

u-erdemir, 2021-12-08:
Yes in theory, its true but there is no IP section in the fpga image. Do you have any plan to integrate it on your product? That will be awesome