plutoplus/plutoplus #18: boot from SD card

idealdelay, 2022-05-20:
What is the suggested process to boot from SD card?

idealdelay, 2022-05-24:
nm, got it to work. 

128 mb fat32 boot partition on sd card. added 'pluto.itb' and 'uenv.txt' to this partition. set jumper sd boot jumper, works fine. 

forced to use the sd card as my images are >>32mb after compiling in gnuradio support for python etc on 2022 buildroot.

#### uenv.txt ###
itb_image=pluto.itb
load_address=0x2080000
sdboot=if mmcinfo; then echo Copying Linux from SD to RAM... && load mmc 0 ${load_address} ${itb_image} && bootm ${load_address} ; fi

F5OEO, 2022-09-22:
in u-boot : mmcinfo reports : sdhci_set_clock: Internal clock never stabilised.
Can you commit or describe your patch ?

fei666888, 2022-09-25:
需要将
Need to connect pin SD-H to 3V3, this will change the MDIO that determines the boot order

DeonMarais64, 2022-10-24:
@idealdelay 
Would you share how you built your image ? I am specifically interested in what you refereed to above "compiling in gnuradio support for python etc on 2022 buildroot".

abusous2000, 2022-11-04:
I wish we have the details showing how to build the SD Card with an image on it

ArielRFF, 2023-06-08:
@idealdelay can you share more details on how you built the SD card ?