plutoplus/plutoplus #19: External Reference Clock

HisenZhang, 2022-06-16:
Hi,

I tried to search for Pluto+ specification online but barely anything was found. This particular question relates to the external clock input. What's the frequency, waveform, and amplitude requirement for such input?

-Hisen

fei666888, 2022-09-25:
The external clock is directly input to AD9363. According to SPEC, the recommended AC signal is 1.3VP-P, the range is 10M-80Mhz, ADI has a method to modify the frequency of REFCLK https://wiki.analog.com/university/ tools/pluto/devs/booting

HisenZhang, 2022-10-05:
> The external clock is directly input to AD9363. According to SPEC, the recommended AC signal is 1.3VP-P, the range is 10M-80Mhz, ADI has a method to modify the frequency of REFCLK https://wiki.analog.com/university/ tools/pluto/devs/booting

Thank you so much! It's much easier to find a good 10MHz GPSDO on the market. Thank you!