plutoplus/plutoplus #20: Bitstream Binaries

hvegh, 2022-08-29:
Dear Howard, PlutoPlus,

Can you publish the bitstreams for version 0.32.1 and the 0.33.3 firmware? 
I would like to experiment with the Linux environment without setting up the Vivado environment...

Much appreciated!

hvegh, 2022-08-30:
Extracted the bitstream from the available firmware images floating around...

[system.bit.v0.32.1.zip](https://github.com/plutoplus/plutoplus/files/9456823/system.bit.v0.32.1.zip)
[system.bit.v0.33.3.zip](https://github.com/plutoplus/plutoplus/files/9456824/system.bit.v0.33.3.zip)

abusous2000, 2022-11-01:
I wonder how to apply the bitstream to Pluto+? I have searched & found no instructions

hvegh, 2022-11-04:
Point to the firmware file:

export HDF_FILE="../firmware/plutosdr-fw-v0.33-3-gd382-dirty.zip"

make sure system_top.bit is in there:

```
Archive:  ../firmware/plutoplus/plutosdr-fw-v0.33-3-gd382-dirty.zip
 Length   Method    Size  Cmpr    Date    Time   CRC-32   Name
--------  ------  ------- ---- ---------- ----- --------  ----
       0  Stored        0   0% 08-30-2022 23:01 00000000  plutosdr-fw-v0.33-3-gd382-dirty/
21185631  Defl:X 20431237   4% 06-18-2021 02:50 ffffffff  plutosdr-fw-v0.33-3-gd382-dirty/pluto.dfu
  651413  Defl:X   251714  61% 06-18-2021 02:50 2f0ec80e  plutosdr-fw-v0.33-3-gd382-dirty/boot.frm
  519300  Defl:X   249050  52% 06-18-2021 02:50 ffffffff  plutosdr-fw-v0.33-3-gd382-dirty/boot.dfu
21185648  Defl:X 20431254   4% 06-18-2021 02:50 f7d4a9dc  plutosdr-fw-v0.33-3-gd382-dirty/pluto.frm
  131088  Defl:X     2333  98% 06-18-2021 02:50 ffffffff  plutosdr-fw-v0.33-3-gd382-dirty/uboot-env.dfu
  965296  Defl:X   387071  60% 08-30-2022 22:59 32061d29  plutosdr-fw-v0.33-3-gd382-dirty/system_top.bit
--------          -------  ---                            -------
44638376         41752659   7%                            7 files

```

and change the Makefile:

```
diff --git a/Makefile b/Makefile
index c0bb761..52d5e0e 100644
--- a/Makefile
+++ b/Makefile
@@ -137,7 +137,7 @@ ifeq (1, ${HAVE_VIVADO})
 else ifneq ($(HDF_FILE),)
        cp $(HDF_FILE) $@
 else ifneq ($(HDF_URL),)
-       wget -T 3 -t 1 -N --directory-prefix build $(HDF_URL)
+       #wget -T 3 -t 1 -N --directory-prefix build $(HDF_URL)
 endif
 
 ### TODO: Build system_top.hdf from src if dl fails - need 2016.2 for that ...
@@ -148,7 +148,7 @@ ifeq (1, ${HAVE_VIVADO})
        bash -c "source $(VIVADO_SETTINGS) && xsdk -batch -source scripts/create_fsbl_project.tcl"
 else
        mkdir -p build/sdk/hw_0
-       unzip -o build/system_top.hdf system_top.bit -d build/sdk/hw_0
+       unzip -j -o build/system_top.hdf *system_top.bit -d build/sdk/hw_0
 endif
 
 build/system_top.bit: build/sdk/hw_0/system_top.bit

```

This works for me, but your mileage may vary. Br