plutoplus/plutoplus #24: Reset device - no ad9361-phy,model found

nmaas87, 2022-11-28:
After trying to load the new 34 firmware as mentioned here: https://github.com/plutoplus/plutoplus/issues/14 - the pluto plus (ver.2) does not show 

        ad9361-phy,xo_correction:
        ad9361-phy,model:

anymore.
Also the device cannot be opened anymore. It is still available via iio_info, but does not work in any SDR program anymore.
How to get it working again?

Library version: 0.24 (git tag: c4498c2)
Compiled with backends: xml ip usb serial
IIO context created with usb backend.
Backend version: 0.21 (git tag: v0.21  )
Backend description string: Linux (none) 4.19.0-g8e9fc41dc7db #25 SMP PREEMPT Mon Sep 28 09:16:27 CST 2020 armv7l
IIO context has 15 attributes:
        hw_model: Analog Devices PlutoSDR Rev.C (Z7010-)
        hw_model_variant: 1
        hw_serial: 10400094c2950017f3ff23001c1c0c64cf
        fw_version: v0.32-1-g7bdc-dirty
        ad9361-phy,xo_correction:
        ad9361-phy,model:
        local,kernel: 4.19.0-g8e9fc41dc7db
        uri: usb:1.4.5
        usb,idVendor: 0456
        usb,idProduct: b673
        usb,release: 2.0
        usb,vendor: Analog Devices Inc.
        usb,product: PlutoSDR (ADALM-PLUTO)
        usb,serial: 10400094c2950017f3ff23001c1c0c64cf
        usb,libusb: 1.0.22.11312

nmaas87, 2022-11-28:
dmesg shows
````
ad9361 spi0.0: ad9361_probe : enter (ad9364)
ad9361 spi0.0: Failed to set BB ref clock rate (-22)
ad9361: probe of spi0.0 failed with error -22
fpga_manager fpga0: Xilinx Zynq FPGA Manager registered
````

so is there something wrong with a setting?

DeonMarais64, 2022-12-01:
Hi,
I assume the above output is from `iio_info -u usb:x.y.z` (x.y.z reported by `iio_info -s` ?
It seems as if the `ad9363` is not responding on the spi.
I have no idea what the problem may be. But I can suggest a few things.

- Are the jumpers correctly set ? https://github.com/plutoplus/plutoplus/issues/14#issue-1048927615
- When you loaded the firmware, did you load both `boot.dfu` and `pluto.dfu` (if you loaded via `dfu-util.exe`) or `boot.frm` and `pluto.frm` (if you loaded via USB-MSD) ? 
- If you loaded via `dfu-util.exe`, then try and load via USB-MSD. I have found it to be is less error prone.

Below is the output I am getting after loading firmware (plutoplus, with ethernet working) I compiled yesterday (I will post the binaries if you like),

```
(base) C:\Users\deon>iio_info -u usb:1.0.5
Library version: 0.24 (git tag: v0.24)
Compiled with backends: xml ip usb
IIO context created with usb backend.
Backend version: 0.24 (git tag: v0.24)
Backend description string: Linux (none) 5.10.0-98231-g9dfba10b795d-dirty #1 SMP PREEMPT Wed Nov 30 15:36:03 SAST 2022 armv7l
IIO context has 15 attributes:
        hw_model: Analog Devices PlutoSDR Rev.C (Z7010-AD9361)
        hw_model_variant: 1
        hw_serial: 1000002313575612002600260311112a3e
        fw_version: v0.35-1-g714c-dirty
        ad9361-phy,xo_correction: 40000000
        ad9361-phy,model: ad9361
        local,kernel: 5.10.0-98231-g9dfba10b795d-dirty
        uri: usb:1.0.5
        usb,idVendor: 0456
        usb,idProduct: b673
        usb,release: 2.0
        usb,vendor: Analog Devices Inc.
        usb,product: PlutoSDR (ADALM-PLUTO)
        usb,serial: 1000002313575612002600260311112a3e
        usb,libusb: 1.0.26.11724
IIO context has 4 devices:
        iio:device0: ad9361-phy
                11 channels found:
                        altvoltage1: TX_LO (output)
                        8 channel-specific attributes found:
                                attr  0: external value: 0
                                attr  1: fastlock_load value: 0
                                attr  2: fastlock_recall ERROR: Invalid argument (22)
                                attr  3: fastlock_save value: 0 255,255,255,255,255,182,255,186,255,255,203,255,255,255,241,84
                                attr  4: fastlock_store value: 0
                                attr  5: frequency value: 435000000
                                attr  6: frequency_available value: [46875001 1 6000000000]
                                attr  7: powerdown value: 0
                        voltage1:  (output)
                        10 channel-specific attributes found:
                                attr  0: filter_fir_en value: 1
                                attr  1: hardwaregain value: -10.000000 dB
                                attr  2: hardwaregain_available value: [-89.750000 0.250000 0.000000]
                                attr  3: rf_bandwidth value: 200000
                                attr  4: rf_bandwidth_available value: [200000 1 40000000]
                                attr  5: rf_port_select value: A
                                attr  6: rf_port_select_available value: A B
                                attr  7: rssi value: 0.00 dB
                                attr  8: sampling_frequency value: 1000000
                                attr  9: sampling_frequency_available value: [520833 1 30720000]
                        voltage0:  (input)
                        15 channel-specific attributes found:
                                attr  0: bb_dc_offset_tracking_en value: 1
                                attr  1: filter_fir_en value: 1
                                attr  2: gain_control_mode value: manual
                                attr  3: gain_control_mode_available value: manual fast_attack slow_attack hybrid
                                attr  4: hardwaregain value: 43.000000 dB
                                attr  5: hardwaregain_available value: [-1 1 73]
                                attr  6: quadrature_tracking_en value: 1
                                attr  7: rf_bandwidth value: 18000000
                                attr  8: rf_bandwidth_available value: [200000 1 56000000]
[... 536 lines cut ...]
                        7 channel-specific attributes found:
                                attr  0: calibbias value: 0
                                attr  1: calibphase value: 0.000000
                                attr  2: calibscale value: 1.000000
                                attr  3: label ERROR: Filename too long (38)
                                attr  4: samples_pps ERROR: No such device (19)
                                attr  5: sampling_frequency value: 1000000
                                attr  6: sampling_frequency_available value: 1000000 125000
                2 device-specific attributes found:
                                attr  0: sync_start_enable value: disarm
                                attr  1: sync_start_enable_available value: arm
                2 buffer-specific attributes found:
                                attr  0: data_available value: 0
                                attr  1: length_align_bytes value: 8
                2 debug attributes found:
                                debug attr  0: pseudorandom_err_check value: CH0 : PN9 : Out of Sync : PN Error
CH1 : PN9 : Out of Sync : PN Error
CH2 : PN9 : Out of Sync : PN Error
CH3 : PN9 : Out of Sync : PN Error
                                debug attr  1: direct_reg_access value: 0x4
                No trigger on this device

(base) C:\Users\deon>
```
I hope this helps.

nmaas87, 2022-12-01:
Hey @DeonMarais64 - thanks for your comment! :)

I assume the above output is from iio_info -u usb:x.y.z (x.y.z reported by iio_info -s ?
 > thats correct
It seems as if the ad9363 is not responding on the spi.
> yeah, I feel that is true...

Are the jumpers correctly set ? https://github.com/plutoplus/plutoplus/issues/14#issue-1048927615
> yes, they are

When you loaded the firmware, did you load both boot.dfu and pluto.dfu (if you loaded via dfu-util.exe) or boot.frm and pluto.frm (if you loaded via USB-MSD) ?
> I tried both the dfu method and the usb-msd but both did not work. When using DFU I also flashed the uboot-env.dfu

If you loaded via dfu-util.exe, then try and load via USB-MSD. I have found it to be is less error prone.
> Sadly tried and did not work

Below is the output I am getting after loading firmware (plutoplus, with ethernet working) I compiled yesterday (I will post the binaries if you like),
> looking very good! I would love to get the binaries and try it with those. You are also using a pluto plus v2?
> Could you (if your time allows and you can post your files) also consider taking a look at #26 ?
> Maybe something else has gone wrong and I tried to write down in issue #26 how to backup a working device with
> all variables. I would then try to flash all your backed up partitions to mine device (assuming its a pluto plus v2)
> and see if this brings it to a working state

> Thanks so much for your time and help!

DeonMarais64, 2022-12-01:
Hi,
I have just recompiled tested and posted,
https://github.com/DeonMarais64/PlutoPlusSDR-FW/releases
This version does not have all the add ons mentioned here https://github.com/plutoplus/plutoplus/releases/tag/v1.33

I will soon be posting one that does.

nmaas87, 2022-12-01:
Thank you, I tried the frm files (both boot.frm and pluto.frm), but sadly this does not resolve the issue.
Maybe if your time allows you could still take a look at #26 and export those files - in case something went wrong with the uboot settings or they were never correctly set from the manufacturer?

````
Library version: 0.24 (git tag: c4498c2)
Compiled with backends: xml ip usb serial
IIO context created with usb backend.
Backend version: 0.24 (git tag: v0.24)
Backend description string: Linux (none) 5.10.0-98231-g9dfba10b795d-dirty #1 SMP PREEMPT Thu Dec 1 13:32:23 SAST 2022 armv7l
IIO context has 15 attributes:
        hw_model: Analog Devices PlutoSDR Rev.C (Z7010-)
        hw_model_variant: 1
        hw_serial: 10400094c2950017f3ff23001c1c0c64cf
        fw_version: v0.35-1-g714c-dirty
        ad9361-phy,xo_correction:
        ad9361-phy,model:
        local,kernel: 5.10.0-98231-g9dfba10b795d-dirty
        uri: usb:1.11.5
        usb,idVendor: 0456
        usb,idProduct: b673
        usb,release: 2.0
        usb,vendor: Analog Devices Inc.
        usb,product: PlutoSDR (ADALM-PLUTO)
        usb,serial: 10400094c2950017f3ff23001c1c0c64cf
        usb,libusb: 1.0.22.11312
IIO context has 1 devices:
        iio:device0: xadc
                10 channels found:
                        voltage5: vccoddr (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 2046
                                attr  1: scale value: 0.732421875
                        voltage0: vccint (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 1399
                                attr  1: scale value: 0.732421875
                        voltage4: vccpaux (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 2461
                                attr  1: scale value: 0.732421875
                        temp0:  (input)
                        3 channel-specific attributes found:
                                attr  0: offset value: -2219
                                attr  1: raw value: 2534
                                attr  2: scale value: 123.040771484
                        voltage7: vrefn (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 5
                                attr  1: scale value: 0.732421875
                        voltage1: vccaux (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 2460
                                attr  1: scale value: 0.732421875
                        voltage2: vccbram (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 1395
                                attr  1: scale value: 0.732421875
                        voltage3: vccpint (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 1396
                                attr  1: scale value: 0.732421875
                        voltage8:  (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 3690
                                attr  1: scale value: 0.244140625
                        voltage6: vrefp (input)
                        2 channel-specific attributes found:
                                attr  0: raw value: 1707
                                attr  1: scale value: 0.732421875
                1 device-specific attributes found:
                                attr  0: sampling_frequency value: 961538
                No trigger on this device
````

nmaas87, 2023-05-30:
See https://github.com/plutoplus/plutoplus/issues/26#issuecomment-1567776189 for resolution

closed 2023-05-30 by nmaas87