plutoplus/plutoplus #26: Backup and recover your device

nmaas87, 2022-11-29:
As I really need a backup from a "virgin" Pluto Plus v2 I am writing this article in hopes someone is so kind to provide the files, thank you! (PS: It will also help you in case something goes wrong and you need to restore...)

0.) Connect your PlutoPlus to your computer via usb and let it boot

1.) Open your webbrowser (http://192.168.2.1) and save the website so that one has the version info

2.) Connect via ssh (192.168.2.1, user root, password analog) to the PlutoPlus

3.) Get all the configured uboot parameters with
````fw_printenv````
- please save the output into a textfile

4.) Reboot it into dfu mode:
````pluto_reboot sf````

5.) Download the dfu-util files from Mr Dirlingers Webpage into a folder:
http://tiger.dirlinger.at/plutoplus/
dfu-util.exe
msvcp120.dll
msvcr120.dll	 

6.) Go into the folder and execute
````.\dfu-util.exe -l````
this should show you all the "partitions" of your pluto:
````
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Cannot open DFU device 0a5c:21e1
Found DFU: [0456:b674] ver=0221, devnum=11, cfg=1, intf=0, path="2-3", alt=4, name="spare.dfu", serial="UNKNOWN"
Found DFU: [0456:b674] ver=0221, devnum=11, cfg=1, intf=0, path="2-3", alt=3, name="uboot-env.dfu", serial="UNKNOWN"
Found DFU: [0456:b674] ver=0221, devnum=11, cfg=1, intf=0, path="2-3", alt=2, name="uboot-extra-env.dfu", serial="UNKNOWN"
Found DFU: [0456:b674] ver=0221, devnum=11, cfg=1, intf=0, path="2-3", alt=1, name="firmware.dfu", serial="UNKNOWN"
Found DFU: [0456:b674] ver=0221, devnum=11, cfg=1, intf=0, path="2-3", alt=0, name="boot.dfu", serial="UNKNOWN"
````
7.) We are now going to download these for a backup to your local computer like this:
(-a is the partiton name, -U means upload to that file - if you repleace -U with -D you will download a local file to the Pluto - if you need to restore it)
````
.\dfu-util.exe -a boot.dfu  -U .\plus-boot.dfu
.\dfu-util.exe -a uboot-env.dfu -U .\plus-uboot-env.dfu
.\dfu-util.exe -a uboot-extra-env.dfu -U .\plus-uboot-extra-env.dfu
.\dfu-util.exe -a spare.dfu  -U .\plus-spare.dfu
.\dfu-util.exe -a firmware.dfu -U .\plus-pluto.dfu
````
8.) Please zip all the dfu files, the saved website and the fw_printenv output and provide them here, thank you for your help!

DeonMarais64, 2022-12-01:
Please see below, I hope you are able to come right

```
deon@i9-dragfr27:~$ ssh root@plutoplus.local
root@plutoplus.local's password: 
Welcome to:
______ _       _              _________________
| ___ \ |     | |         _  /  ___|  _  \ ___ \
| |_/ / |_   _| |_ ___  _| |_\ `--.| | | | |_/ /
|  __/| | | | | __/ _ \|_   _|`--. \ | | |    /
| |   | | |_| | || (_) | |_| /\__/ / |/ /| |\ \
\_|   |_|\__,_|\__\___/      \____/|___/ \_| \_|

v0.35-1-g714c-dirty
https://github.com/DeonMarais64/PlutoPlusSDR-FW
# printenv
USER=root
SSH_CLIENT=192.168.3.10 46670 22
SHLVL=1
HOME=/root
SSH_TTY=/dev/pts/0
PS1=# 
LOGNAME=root
TERM=xterm-256color
PATH=/bin:/sbin:/usr/bin:/usr/sbin
SHELL=/bin/sh
PWD=/root
SSH_CONNECTION=192.168.3.10 46670 192.168.3.1 22
EDITOR=/bin/vi
# fw_printenv
adi_loadvals=fdt addr ${fit_load_address} && fdt get value fdt_choosen /configurations/${fit_config}/ fdt && fdt get addr fdtaddr /images/${fdt_choosen} data && fdt addr ${fdtaddr}; fdt get value model / model; if test "${model}" > "Analog Devices Pluto"; then run adi_loadvals_pluto; fi; 
adi_loadvals_pluto=if test -n "${ad936x_ext_refclk}" && test ! -n "${ad936x_skip_ext_refclk}"; then fdt set /clocks/clock@0 clock-frequency ${ad936x_ext_refclk}; fi; if test -n "${ad936x_ext_refclk_override}"; then fdt set /clocks/clock@0 clock-frequency ${ad936x_ext_refclk_override}; fi; if test -n "${refclk_source}" && test ! "${refclk_source}" = "internal" && test ! "${refclk_source}" = "external"; then setenv refclk_source internal; saveenv; fi; if test "${refclk_source}" = "internal" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)" ; then fdt rm /amba/gpio@e000a000/clock_extern_en || fdt rm /axi/gpio@e000a000/clock_extern_en; fi; if test "${refclk_source}" = "external" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)" ; then fdt rm /amba/gpio@e000a000/clock_internal_en || fdt rm /axi/gpio@e000a000/clock_internal_en; fi; if test  "${attr_val}" = "ad9361" && test ! "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)" ; then setenv attr_val ad9363a; saveenv; fi; if test -n "${attr_val}" && test ! "${attr_val}" = "ad9361" && test ! "${attr_val}" = "ad9363a" && test ! "${attr_val}" = "ad9364"; then setenv attr_val ad9363a; saveenv; fi; if test -n "${mode}" && test ! "${mode}" = "1r1t" && test ! "${mode}" = "2r2t"; then setenv mode 1r1t; saveenv; fi; if test -n "${attr_name}" && test -n "${attr_val}"; then fdt set /amba/spi@e0006000/ad9361-phy@0 ${attr_name} ${attr_val} || fdt set /axi/spi@e0006000/ad9361-phy@0 ${attr_name} ${attr_val}; fi; if test "${mode}" = "1r1t" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)"; then fdt rm /amba/spi@e0006000/ad9361-phy@0 adi,2rx-2tx-mode-enable || fdt rm /axi/spi@e0006000/ad9361-phy@0 adi,2rx-2tx-mode-enable; fdt set /fpga-axi/cf-ad9361-dds-core-lpc@79024000 compatible adi,axi-ad9364-dds-6.00.a; fi; if test -n "${cs_gpio}" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)"; then fdt set /amba/axi_quad_spi@7C430000/ cs-gpios "<0x06 ${cs_gpio} 0>" || fdt set /axi/axi_quad_spi@7C430000/ cs-gpios "<0x06 ${cs_gpio} 0>"; fi; if test -n "${attr_val}" && test "${attr_val}" = "ad9364"; then fdt set /fpga-axi/cf-ad9361-dds-core-lpc@79024000 compatible adi,axi-ad9364-dds-6.00.a; if test ! "${mode}" = "1r1t"; then fdt rm /amba/spi@e0006000/ad9361-phy@0 adi,2rx-2tx-mode-enable || fdt rm /axi/spi@e0006000/ad9361-phy@0 adi,2rx-2tx-mode-enable; setenv mode 1r1t; saveenv; fi; fi; 
baudrate=115200
bitstream_image=system.bit.bin
bitstream_size=0x400000
bootcmd=run $modeboot
bootdelay=3
bootenv=uEnv.txt
boot_image=BOOT.bin
boot_size=0xF00000
clear_reset_cause=mw f8000008 df0d && mw f8000258 00400000 && mw f8000004 767b
devicetree_image=devicetree.dtb
devicetree_load_address=0x2000000
devicetree_size=0x20000
dfu_mmc_info=set dfu_alt_info ${kernel_image} fat 0 1\\;${devicetree_image} fat 0 1\\;${ramdisk_image} fat 0 1
dfu_mmc=run dfu_mmc_info && dfu 0 mmc 0
dfu_ram=echo Entering DFU RAM mode ... && run dfu_ram_info && dfu 0 ram 0
dfu_ram_info=set dfu_alt_info dummy.dfu ram 0 0\\;firmware.dfu ram ${fit_load_address} 0x1E00000
dfu_sf=gpio set 15;set stdout serial@e0001000;echo Entering DFU SF mode ... && run dfu_sf_info && dfu 0 sf 0:0:40000000:0 && if test -n ${dfu_alt_num} && test ${dfu_alt_num} = 1; then set fit_size ${filesize} && set dfu_alt_num && env save; fi;gpio clear 15;
dfu_sf_info=set dfu_alt_info boot.dfu raw 0x0 0x100000\\;firmware.dfu raw 0x200000 0x1E00000\\;uboot-extra-env.dfu raw 0xFF000 0x1000\\;uboot-env.dfu raw 0x100000 0x20000\\;spare.dfu raw 0x120000 0xE0000
ethaddr=00:0a:35:00:01:22
extraenv_load_address=0x207E000
fdt_high=0x20000000
fit_config=config@0
fit_load_address=0x2080000
importbootenv=echo Importing environment from SD ...; env import -t ${loadbootenv_addr} $filesize
initrd_high=0x20000000
jtagboot=env default -a;sf probe && sf protect unlock 0 100000 && run dfu_sf; 
kernel_image=uImage
loadbit_addr=0x100000
loadbootenv_addr=0x2000000
loadbootenv=load mmc 0 ${loadbootenv_addr} ${bootenv}
maxcpus=2
mode=2r2t
netmask=255.255.255.0
preboot=
preboot=if test $modeboot = sdboot && env run sd_uEnvtxt_existence_test; then if env run loadbootenv; then env run importbootenv; fi; fi; 
qspiboot_extraenv=sf read ${extraenv_load_address} 0xFF000 0x1000 && env import -c ${extraenv_load_address} 0x1000 || true 
[... cut ...]
  312 avahi    avahi-daemon: running [plutoplus.local]
  320 root     /usr/sbin/dropbear -R
  323 root     input-event-daemon
  324 root     /sbin/getty -L ttyPS0 0 vt100
  964 root     [kworker/1:0-eve]
 1332 root     [kworker/u4:2-ev]
 1590 root     [kworker/1:1-eve]
 1738 root     /usr/sbin/dropbear -R
 1745 root     /sbin/getty -L ttyGS0 0 vt100
 1746 root     -sh
 1827 root     sleep 1
 1828 root     ps aux
# 
```
[PlutoPlus.zip](https://github.com/plutoplus/plutoplus/files/10132936/PlutoPlus.zip)

nmaas87, 2022-12-01:
Thank you so much, I'll definitely give it a shot tonight!
And I think the community will be really happy to see a new firmware for their plutoplus - thank you for being so kind! :)

DeonMarais64, 2022-12-01:
I also loaded this firmware as you originally asked https://github.com/plutoplus/plutoplus/releases/tag/v1.33.
Please see below. Also kindly let me know how it turns out.
```
(base) C:\Programs\Radio\PlutoSDR-M2K\pp_1.33\PlutoSDR>ssh root@pluto.local
root@pluto.local's password:
Welcome to:
 ____  _       _
|  _ \| |_   _| |_ ___    _
| |_) | | | | | __/ _ \ _| |_
|  __/| | |_| | || (_) |_   _|
|_|   |_|\__,_|\__\___/  |_|

v0.33-3-gd382-dirty
https://github.com/plutoplus/plutoplus
# printenv
USER=root
SSH_CLIENT=192.168.2.10 62076 22
SHLVL=1
HOME=/root
SSH_TTY=/dev/pts/0
PS1=#
LOGNAME=root
TERM=xterm-256color
PATH=/bin:/sbin:/usr/bin:/usr/sbin
SHELL=/bin/sh
PWD=/root
SSH_CONNECTION=192.168.2.10 62076 192.168.2.1 22
EDITOR=/bin/vi
# fw_printenv
adi_loadvals=fdt addr ${fit_load_address} && fdt get value fdt_choosen /configurations/${fit_config}/ fdt && fdt get addr fdtaddr /images/${fdt_choosen} data && fdt addr ${fdtaddr}; fdt get value model / model; if test "${model}" > "Analog Devices Pluto"; then run adi_loadvals_pluto; fi;
adi_loadvals_pluto=if test -n "${ad936x_ext_refclk}" && test ! -n "${ad936x_skip_ext_refclk}"; then fdt set /clocks/clock@0 clock-frequency ${ad936x_ext_refclk}; fi; if test -n "${ad936x_ext_refclk_override}"; then fdt set /clocks/clock@0 clock-frequency ${ad936x_ext_refclk_override}; fi; if test -n "${refclk_source}" && test ! "${refclk_source}" = "internal" && test ! "${refclk_source}" = "external"; then setenv refclk_source internal; saveenv; fi; if test "${refclk_source}" = "internal" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)" ; then fdt rm /amba/gpio@e000a000/clock_extern_en; fi; if test "${refclk_source}" = "external" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)" ; then fdt rm /amba/gpio@e000a000/clock_internal_en; fi; if test  "${attr_val}" = "ad9361" && test ! "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)" ; then setenv attr_val ad9363a; saveenv; fi; if test -n "${attr_val}" && test ! "${attr_val}" = "ad9361" && test ! "${attr_val}" = "ad9363a" && test ! "${attr_val}" = "ad9364"; then setenv attr_val ad9363a; saveenv; fi; if test -n "${mode}" && test ! "${mode}" = "1r1t" && test ! "${mode}" = "2r2t"; then setenv mode 1r1t; saveenv; fi; if test -n "${attr_name}" && test -n "${attr_val}"; then fdt set /amba/spi@e0006000/ad9361-phy@0 ${attr_name} ${attr_val}; fi; if test "${mode}" = "1r1t" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)"; then fdt rm /amba/spi@e0006000/ad9361-phy@0 adi,2rx-2tx-mode-enable; fdt set /fpga-axi/cf-ad9361-dds-core-lpc@79024000 compatible adi,axi-ad9364-dds-6.00.a; fi; if test -n "${cs_gpio}" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)"; then fdt set /amba/axi_quad_spi@7C430000/ cs-gpios "<0x06 ${cs_gpio} 0>"; fi; if test -n "${attr_val}" && test "${attr_val}" = "ad9364"; then fdt set /fpga-axi/cf-ad9361-dds-core-lpc@79024000 compatible adi,axi-ad9364-dds-6.00.a; if test ! "${mode}" = "1r1t"; then fdt rm /amba/spi@e0006000/ad9361-phy@0 adi,2rx-2tx-mode-enable; setenv mode 1r1t; saveenv; fi; fi;
baudrate=115200
bitstream_image=system.bit.bin
bitstream_size=0x400000
bootcmd=run $modeboot
bootdelay=3
bootenv=uEnv.txt
boot_image=BOOT.bin
boot_size=0xF00000
clear_reset_cause=mw f8000008 df0d && mw f8000258 00400000 && mw f8000004 767b
devicetree_image=devicetree.dtb
devicetree_load_address=0x2000000
devicetree_size=0x20000
dfu_mmc_info=set dfu_alt_info ${kernel_image} fat 0 1\\;${devicetree_image} fat 0 1\\;${ramdisk_image} fat 0 1
dfu_mmc=run dfu_mmc_info && dfu 0 mmc 0
dfu_ram=echo Entering DFU RAM mode ... && run dfu_ram_info && dfu 0 ram 0
dfu_ram_info=set dfu_alt_info dummy.dfu ram 0 0\\;firmware.dfu ram ${fit_load_address} 0x1E00000
dfu_sf=gpio set 15;set stdout serial@e0001000;echo Entering DFU SF mode ... && run dfu_sf_info && dfu 0 sf 0:0:40000000:0 && if test -n ${dfu_alt_num} && test ${dfu_alt_num} = 1; then set fit_size ${filesize} && set dfu_alt_num && env save; fi;gpio clear 15;
dfu_sf_info=set dfu_alt_info boot.dfu raw 0x0 0x100000\\;firmware.dfu raw 0x200000 0x1E00000\\;uboot-extra-env.dfu raw 0xFF000 0x1000\\;uboot-env.dfu raw 0x100000 0x20000\\;spare.dfu raw 0x120000 0xE0000
ethaddr=00:0a:35:00:01:22
extraenv_load_address=0x207E000
fdt_high=0x20000000
fit_config=config@0
fit_load_address=0x2080000
fit_size=0x900000
importbootenv=echo Importing environment from SD ...; env import -t ${loadbootenv_addr} $filesize
initrd_high=0x20000000
ipaddr=192.168.2.1
ipaddr_host=192.168.2.10
jtagboot=env default -a;sf probe && sf protect unlock 0 100000 && run dfu_sf;
kernel_image=uImage
loadbit_addr=0x100000
loadbootenv_addr=0x2000000
loadbootenv=load mmc 0 ${loadbootenv_addr} ${bootenv}
maxcpus=2
mode=2r2t
netmask=255.255.255.0
preboot=
preboot=if test $modeboot = sdboot && env run sd_uEnvtxt_existence_test; then if env run loadbootenv; then env run importbootenv; fi; fi;
qspiboot_extraenv=sf read ${extraenv_load_address} 0xFF000 0x1000 && env import -c ${extraenv_load_address} 0x1000 || true
[... 181 lines cut ...]

(base) C:\Programs\Radio\PlutoSDR-M2K\pp_1.33\PlutoSDR>..\..\dfu-util.exe -a firmware.dfu -U .\plus-pluto.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 0456:b674
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #1 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 4096
Copying data from DFU device to PC
Upload  [=========================] 100%     31457280 bytes
Upload done.

(base) C:\Programs\Radio\PlutoSDR-M2K\pp_1.33\PlutoSDR>
```
[PlutoSDR.zip](https://github.com/plutoplus/plutoplus/files/10134578/PlutoSDR.zip)

nmaas87, 2022-12-01:
Thank you very much @DeonMarais64 ! :) 
I could restore your 0.35+ firmware on my pluto plus - sadly it did not help.
I could see that everything was on there - but it did not help.
Thinking about it, I don't know if buying a xilinx programmer could help to directly program the FPGA, but I don't know if this would make a difference or if everything that can be flashed will be flashed by DFU (via USB) or USB-MSD flashing.
It really seems like my PlutoPlus was DoA and will not come back from this state...
I will see that I flash the 0.33/1.33er backup of yours as well and see if this works, but chances seem slim now that I also can rule out any "trash" writen in the other partitions. But thanks again, that was really awesome to at least try to get it working.

nmaas87, 2022-12-02:
Also tested the 0.33/1.33 files - sadly also no joy.
I will see wheter it makes sense to get a programmer, but I think this might not help as I fear the main chip is just defect...

DeonMarais64, 2022-12-02:
Thanks for responding, and it is indeed a pity that you have not been able to bring your unit back.

For reference and to everyone's benefit would you mind confirming how your device ended up this way,

From your description at the [`beginning of this thread`](https://github.com/plutoplus/plutoplus/issues/24#issue-1467185375) I gather in summery,
- You had a fully functional `Pluto+` as shipped, probably running [`releases/v1.33`](https://github.com/plutoplus/plutoplus/releases/tag/v1.33),
- You then simply upgraded to [this firmware](http://tiger.dirlinger.at/plutoplus/34plus/) (`pluto.dfu`, `boot.dfu` and `boot-env.dfu`) via the [`DFU Update`](https://wiki.analog.com/university/tools/pluto/users/firmware#dfu_update) process,
- Subsequently the unit does not respond to anything that relates to the `ad9363` so to all purposes as an SDR your `Pluto+` is non functional,
- Any attempts to restore its working state (by loading original and/or any other firmware) have thus far been unsuccessful.

I have previously loaded http://tiger.dirlinger.at/plutoplus/34plus/ (albeit via [`USB-MSD`](https://wiki.analog.com/university/tools/pluto/users/firmware#mass_storage_update) process) and confirm that it does work and caused my unit no harm. I have also loaded various other firmware's this way and have so far not had any problems.

For good measure, maybe you should try loading the latest [ADI Official](https://github.com/analogdevicesinc/plutosdr-fw/releases) firmware and setting the [Pluto+ Jumper](https://github.com/plutoplus/plutoplus#jumpers-and-pinouts) appropriately.

nmaas87, 2022-12-02:
I had a Pluto+ v.2.0 shipped, however, it was my first Pluto and I did not know yet that it was broken. I only saw the xadc - and it did not work in any software. So I flashed it from 1.32 (not available anywhere...) - which did neither improve nor worsen the problem. But then I had the oppurtunity to buy a real Pluto used and figured out that the + was basically DOA, because I saw the ad9363, more IIO devices and that one was working.

That was the original shipping configuration (1.32), however I never found it online, the manufacture does not send it out and I did not backup it because I did not yet know how to do that - and thought I would improve the situation anyway with going to 1.33 - as this version was also mentioned in the ZIP file the manufacturer send me. I also tried 0.31, 0.32, 0.33 and 0.35 of the original software - did sadly not help.

````
Model | Analog Devices PlutoSDR Rev.C (Z7010-)
Build | v0.32-1-g7bdc-dirty
Linux | Linux pluto 4.19.0-g8e9fc41dc7db #25 SMP PREEMPT Mon Sep 28 09:16:27 CST 2020 armv7l GNU-Linux;2 core(s)
U-Boot | U-Boot PlutoSDR v0.20-PlutoSDR-00053-g89d0754 (May 28 2019 - 19:24:22 -0700)
FPGA | dev_prj_2018_r1-935-gd54c2
Root FS | v0.20-13876-g1daa6-dirty
IIO | Library version: 0.21 (git tag: v0.21)
````

DeonMarais64, 2022-12-02:
Ok, so it was a DOA situation then and not a failure due to loading of firmware or something of that nature,

BTW,
In case you need or want `plutosdr-fw-v0.32-1-g7bdc-dirty.zip`
From https://drive.google.com/file/d/1OnJOtojzaN82d3yuHzG1YkfeXNFKKYbA/view
[plutosdr-fw-v0.32-1-g7bdc-dirty.zip](https://github.com/plutoplus/plutoplus/files/10138125/plutosdr-fw-v0.32-1-g7bdc-dirty.zip)

nmaas87, 2022-12-02:
Thanks a lot, I tried this as well - but sadly did not work.
Yeah, I think maybe the chip itself is defective - the FPGA is not correct written - or the crystal could have a problem...
Well, lets see if I can do something about it, but its looking a bit grim.

DeonMarais64, 2022-12-02:
I hope I not being an nuisance but this has me intrigued,

The schematics are from from https://wiki.analog.com/university/tools/pluto/hacking/hardware

I think it is probably more likely that  the `ad9363` is faulty, as the `spi` interface to/from the `ad9363` should only rely on the clock issued by the `XC7Z010` and not any other clock or crystal source.

Unfortunately the `Pluto+` schematics are terrible but the `Pluto+` most closely resembles that for `Rev-B`. The `Rev-D` schematic is actually that for `Rev-C` of the `Pluto`

```
IO_L24_35_SPI_DI
IO_L24_35_SPI_DO
IO_L23_35_SPI_ENB
IO_L23_35_SPI_CLK
```

I think the `iio_xxx` commands from the `ADI` library does allow you to prole the `ad9363` registers directly, it would be interesting to see what you can find there.

[plutosdr_schematic_reva.pdf](https://github.com/plutoplus/plutoplus/files/10138609/plutosdr_schematic_reva.pdf)
[plutosdr_schematic_revb.pdf](https://github.com/plutoplus/plutoplus/files/10138611/plutosdr_schematic_revb.pdf)
[plutosdr_schematic_revd_0.1.pdf](https://github.com/plutoplus/plutoplus/files/10138612/plutosdr_schematic_revd_0.1.pdf)

nmaas87, 2022-12-02:
Thats a good idea, however, as the ````ad9363```` is not recognized by the Pluto+ this does not work - I cannot probe the ````ad9363```` as the IIO needs to have the SPI device recognized and its drivers loaded before it will show up. But as its not recognized there is no IIO context and hence I cannot use the iio tools. 

````
iio_attr -c ad9361-phy altvoltage0 frequency 2450000000
iio_attr: Error : could not find device (ad9361-phy)
````

PeterBeijer, 2023-05-28:
@nmaas87 @DeonMarais64 
This is an interesting conversation. I'll tell you my journey:

I had a flawless working Pluto +. Upgraded firmware a couple of times, no problem. At some point did another upgrade and the device started to behave intermittently strange.  Found out the AD9361 sometimes was not recognized. 

probing with $ iio_attr -C -u usb:x.x.x 
"Model | Analog Devices PlutoSDR Rev.B (Z7010- ) " Says it all. It should read 
"Model | Analog Devices PlutoSDR Rev.B (Z7010- AD9361) "

After playing with some updates with various firmwares I  finally got it working again. However I found out that both receivers where not working.

I could measure pwr output at both TX channels. Used Satsagen generator for that and measured with my ADALM Pluto (Have 2 Pluto's) and HP power meter. 

So it was time to play with some FW updates again. Surprise: the AD9361 is now no longer recognized again.

# > dmesg shows SPI timeouts. 

I measured all necessary voltages for the AD9361.  They're all OK.
I measured the 40MHz. It 's OK

So . . .  Some conclusions:
1) Either the HW is flaky, which I do not believe, or
2) the FPGA is screwed a bit due to improper handling from my side with pwr on/off in the heat of the game ;-) 

I've seen somewhere a way to use the JTAG stuf to re-flash the FPGA (wish I remember where). That could probably solve the problem. 

Any thoughts?

Peter

DeonMarais64, 2023-05-28:
Hi Peter,

I have not tried loading the ADI firmware since v0.35 (v0.36&v0.37) so I cannot comment on their behavior on the `Pluto+`.
Have you tried loading any of  [`these`](https://github.com/DeonMarais64/PlutoPlusSDR-FW/releases) firmware's ?
With any of them loaded and then using the commands below described [`here`](https://wiki.analog.com/university/tools/pluto/users/customizing#updating_to_the_ad9364) to hack the `ad9364` to an `ad9361`,
```
# fw_setenv attr_name compatible
# fw_setenv attr_val ad9361
# fw_setenv compatible ad9361
# fw_setenv mode 2r2t
# reboot
```
You should get something similar to what is below if your `Pluto+` is fully functional,

```
C:\Users\deon>iio_attr -C -u ip:plutoplus.local
IIO context with 9 attributes:
hw_model: Analog Devices PlutoSDR Rev.C (Z7010-AD9361)
hw_model_variant: 1
hw_serial: 1000002313575612002600260311112a3e
fw_version: v0.35-1-g714c-dirty
ad9361-phy,xo_correction: 40000000
ad9361-phy,model: ad9361
local,kernel: 5.10.0-98231-g9dfba10b795d-dirty
uri: ip:plutoplus.local
ip,ip-addr: 192.168.1.40
```
I have not tried to load the FPGA using `JTag`, but I doubt it would change anything.

I hope this helps.

PeterBeijer, 2023-05-29:
Thank you Deon. I've tried all that. Various versions, DFU, MSD etc. No luck. Investigating now the JTag route. 

Btw, your version numbering on your[ repo here](https://github.com/DeonMarais64/PlutoPlusSDR-FW/releases) is confusing. It has the same version number but different content. . .

Appreciate th work you do though!

cheers,
Peter

DeonMarais64, 2023-05-30:
> Btw, your version numbering on your[ repo here](https://github.com/DeonMarais64/PlutoPlusSDR-FW/releases) is confusing

The reported version number comes from the base ADI repo [`makefile`](https://github.com/analogdevicesinc/plutosdr-fw/blob/714cd8aaeadfa30aecd6d8235d5076a3f7b8e5e3/Makefile#LL24C1-L24C65) which is used for building the firmware.
So this will be the what is reported when running `git describe --abbrev=4 --dirty --always --tags` from the root directory in the post scripted cloned ADI git repo which is where build process starts.

The differences in the versions I posted are described in the release notes. 
Briefly, 
1. [Vanilla ADI FW with ethernet support.](https://github.com/DeonMarais64/PlutoPlusSDR-FW/releases/tag/v0.35%2B)
2. [Identical to 1 above AND additional packages required for DATV.](https://github.com/DeonMarais64/PlutoPlusSDR-FW/releases/tag/v0.35%2B_datv_packages)
3. [Identical to 2 above AND the ability to run user scripts at startup.](https://github.com/DeonMarais64/PlutoPlusSDR-FW/releases/tag/v0.35%2B_datv_packages_autorun)

nmaas87, 2023-05-30:
Thats all I got on notes - sorry cannot provide more support.

The issue was not the software.

I desoldered the EMV/metal cap of the AD936x ADC/DAC and tried reflowing it with a heatgun, but this did not help.
I thought the chip was probably dead as I only got following errors on bootup in dmesg:

````
ad9361 spi0.0: ad9361_probe : enter (ad9363a)
ad9361 spi0.0: ad9361_probe : Unsupported PRODUCT_ID 0xFF
````

So I ordered a new AD936x and thought - yeah, its BGA, but I wanted to try soldering these anyway and the Pluto+ is broken, so what do I got to lose? However - waiting for the chip which has not yet arrived - I still could not stop thinking about the oscilator.
I do not have an oscilioscope and with that no chance of testing the internal clock source, but I thought of something different.

I got myself an Lattice iCEstick FPGA and have been learning to work with this thing a bit.
Actually it got an 12 MHz onboard crystal and an phase-locked loop/PLL with which it can easily generate clocks up till the GHz area. Probably not perfectly stable, but it could. There was an extremly helpful guide on Youtube how to set this up ( https://www.youtube.com/watch?v=gmaSjyUij9E ) - and so I programmed the FPGA to output 42 MHz on a pin (40 Mhz would be like 39.5 MHz due to hardware limitations/divisors in the PLL, so I thought, 42 would be more clean and I can set it in the Pluto...) - and soldered myself an IPEX adapter cable for getting GND and "Clock" to the external clock/CLK-IN input of the Pluto+. I also needed to bridge the EXCLK jumper with GND on the Pluto+. After powering on the FPGA and Pluto+, I set the Pluto+ to use the external clock source at 42 MHz:

````
fw_setenv ad936x_ext_refclk "<42000000>"
fw_setenv refclk_source external
pluto_reboot reset
````

After the reset it was working again:

````
ad9361 spi0.0: ad9361_probe : enter (ad9363a)
random: fast init done
ad9361 spi0.0: ad9361_probe : AD936x Rev 2 successfully initialized
````

I flashed it from the original 0.32+ to the latest 0.35+ DATV firmware - and tried it in SDR Console - it worked again.
Now the fun part: I looked into the config after the upgrade - and it was running on the INTERNAL clock.
I removed the external clock signal as well as the jumper (GND/EXCLK), rebooted - and it was still working.

I cannot say what the issue was - other that I probably poured a weeks of work into reviving this board. (Could be that there was some dirt in the connector which I could not see which made a short circuit and did pull the internal clock down... Or that the AD936x was happy after receiving a clock again and decided to work again...) - I don't know.

[image]

PeterBeijer, 2023-05-30:
This is truly amazing Nico. Thank you for your informative response. I'll dig into it using your info and report back.