plutoplus/plutoplus #32: Pluto+ not recognised, green LED constantly on

kconkas, 2023-04-23:
Hi,

I hope someone can help me with this issue. My plutoplus used to work fine until I plugged it into a computer with seemingly insufficient USB power.

Now when I plug it to my Linux box on which it previously worked, the green LED is constantly on, the blue LED is off, and I can't pluto+ it in the output of `lsusb`.

When I try to boot it in DFU mode by following the instructions in https://github.com/plutoplus/plutoplus#how-to-enter-dfu I get the following:
```
[ 6158.518718] usb 2-1-port5: Cannot enable. Maybe the USB cable is bad?
[ 6158.582515] usb 2-1.5: new low-speed USB device number 93 using ehci-pci
[ 6159.422706] usb 2-1-port5: Cannot enable. Maybe the USB cable is bad?
[ 6159.422891] usb 2-1-port5: attempt power cycle
[ 6160.574704] usb 2-1-port5: Cannot enable. Maybe the USB cable is bad?
[ 6160.638516] usb 2-1.5: new full-speed USB device number 95 using ehci-pci
[ 6161.054382] usb 2-1.5: device not accepting address 95, error -32
[ 6161.054479] usb 2-1-port5: unable to enumerate USB device
```
I tried to plug it into a powered USB hub, to no avail.

Can anyone give me any pointers as to where to go next in terms of troubleshooting?

simbaaz, 2023-05-10:
Same issue here (I unplugged +5V USB before unplugging DATA USB, and now USB enumeration always fails), please help to troubleshoot the issue.

mx0xuu, 2023-07-04:
Same here  - green LED is ON and not detected anymore.
Also, the schematic (PDF) is missing some fonts and not able to use. Anyone got the schematic ?

andyd666, 2023-08-16:
The same. Initially, I connected only data USB. It booted and also opened internal memory with config.txt and other stuff. I also connected via Putty. Then I also attached a power USB and in about a minute it disconnected, Putty said that it lost connection. After that, I couldn't connect again. I don't think the problem is power loss, because I have 5V3A power adapter and still only single green LED is on

OZ1CT-Ben, 2024-06-21:
Hi all...same problem here, some guys by a JTAG programmer and then it is up and run...is that true ???? if yes, where to buy ???
or is there a other solution for this problem ???

video link of my board: https://photos.app.goo.gl/S7QQJ2GXQVRNk89TA

73" Ben

VioletGiraffe, 2024-11-03:
Same problem, might be because I used the power switch on my USB hub to toggle the device off and back on, I never had a problem doing that with any devices, including a couple SDRs...