plutoplus/plutoplus #35: Booting Pluto+ from SD card

ArielRFF, 2023-06-08:
I am working on a Pluto Firmware version which is not mature enough to flash inside my Pluto+.
I am trying (unsuccesfull by the moment) to boot my firmware from the SD.  

The pins SD-H and 3V3 are jumpered together, as seen in the instructions (my Pluto+ version is 3V3 and not 1V8).
Green LED turns on when I connect power, but it stays that way after connecting the USB Data port.
If I disconnect the jumper and power cycle, it boots from internal Flash.

Could someone kindly give me steb-by-step instructions to boot from the SD ?

Shawn-McSorley, 2024-07-02:
I've made some progress in this regard.

It required modifying the u-boot sources a bit. I found most of the changes on a libre-sdr (another pluto clone) github repo. https://github.com/hz12opensource/libresdr.

I think the main change required is

Uboot b/include/configs/zynq-common.h

+	"sdboot=if mmcinfo; then " \
+			"run uenvboot; " \
+			"echo Copying Linux from SD to RAM... && " \
+			"load mmc 0 ${fit_load_address} ${kernel_image} && " \
+			"load mmc 0 ${devicetree_load_address} ${devicetree_image} && " \
+			"load mmc 0 ${ramdisk_load_address} ${ramdisk_image} && " \
+			"bootm ${fit_load_address} ${ramdisk_load_address} ${devicetree_load_address}; " \
+		"fi\0" \

This instructs u-boot to take the firmware on the sd card and load it to flash. 

There are some changes as well required in the makefile to compile a uImage. The main process is

SDIMGDIR = $(CURDIR)/build_sdimg
sdimg: build/
	mkdir $(SDIMGDIR)
	cp build/sdk/fsbl/Release/fsbl.elf 	$(SDIMGDIR)/fsbl.elf  
	cp build/sdk/system_top/hw/system_top.bit 	$(SDIMGDIR)/system_top.bit
	cp build/u-boot.elf 			$(SDIMGDIR)/u-boot.elf
	cp $(CURDIR)/linux/arch/arm/boot/uImage	$(SDIMGDIR)/uImage
	cp build/zynq-$(TARGET)-sdr-revc.dtb 	$(SDIMGDIR)/devicetree.dtb
	cp build/uboot-env.txt  		$(SDIMGDIR)/uEnv.txt
	cp build/rootfs.cpio.gz  		$(SDIMGDIR)/ramdisk.image.gz
	mkimage -A arm -T ramdisk -C gzip -d $(SDIMGDIR)/ramdisk.image.gz $(SDIMGDIR)/uramdisk.image.gz
	touch 	$(SDIMGDIR)/boot.bif
	echo "img : { [bootloader] $(SDIMGDIR)/fsbl.elf $(SDIMGDIR)/system_top.bit  $(SDIMGDIR)/u-boot.elf}" > $(SDIMGDIR)/boot.bif
	bash -c "source $(VIVADO_SETTINGS) && bootgen -image $(SDIMGDIR)/boot.bif -o i $(SDIMGDIR)/BOOT.bin"
	
I've managed to get it booting from the sd card, however it's not 100% yet. Currently it's not loading the bitstream image which is annoying. 

Another issue I've found is the device trees. To get the data port working, I need to load the zynq-pluto-sdr.dtb device tree, and then it boots fine and works like normal (except no ethernet cause it's a stock pluto dtb I think). If I load the rev-c, I get ethernet back and can boot when plugged in to the 5V-in port. But, no bitstream. Still a bit to debug.

Shawn-McSorley, 2024-07-03:
Just an update for anyone hoping to boot from the sd card, I applied the fixes from this  other pluto like repo [Patch Notes](https://github.com/MicroPhase/antsdr-fw/commit/9e5a2d01a084473b6fb151003cb1d4fe08421d34#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5), and it boots from the sd card.

Up next would be partitioning the sd card in to a FAT32 boot partition, and a rootfs ext4 partition with the root filesystem.

F5OEO, 2024-08-28:
@Shawn-McSorley : I am stuck at booting , and stop at zynq-pinctrl 700.pinctrl: zynq pinctrl initialized. I suspect bitstream not loaded. An other point : even with the modified adi_loadval, I have some error regarding fdt. What solution you think solve the issue of the bitstream not loaded ?

Shawn-McSorley, 2024-08-30:
One thing I did find. It's better not to modify the u-boot config during the build process. That is, leave the defaults there. Then modify the generated uboot-env.txt directly. I found that even when booting from the sd card, the u-boot environment that gets loaded is what is in the flash. So if that's modified, the device tree must get corrupted or something, which led to me not seeing the ADI devices or my bitstream modifications.

I didn't find a fix here. I was able to change the firmware and bitstream via sd card booting, just not the u-boot. The u-boot ended up being taken from the flash. But the uboot-env.txt on the sd card still needs to be modified to boot. It was strange.

I've mostly steered away from sd card booting as it was quite a time sink, and I'm more focused on custom hardware in the FPGA.

So just to reiterate, don't modify the uenv that is apart of the build process. Leave it as default. Once it's built, load the firmware to flash like normal. Then, modify the generated uboot-env.txt with the modifcations I linked above. 

Put everything on the sd card, change the jumper, and it should boot from the sd card. I verified it  changing the login banner and rebuilding the firmware, or you can add in something to the FPGA. 

Lastly, if you run fw_printenv once booted from the sd card, you will probably see that it's the default uboot-env. Which is what I encountered. I was hoping to be able to modify the u-boot env on the fly and load changes via the sd card, but I didn't figure that out.