plutoplus/plutoplus #43: Cannot extend range to 70-6000 MHz (update to AD9364/AD9364) - U-Boot env is not used/corrupted

belashou, 2024-07-24:
Hi, could you please assist with the following issue or provide hints what to try next?

### Problem
Cannot extend SDR range with [official instructions from AD]( https://wiki.analog.com/university/tools/pluto/users/customizing). I tried a few firmawares and flashing methods (described below) - the pattern is pretty similar in every case. There is just noise below 300MHz and ADC in `iio_info` is always the same - `ad9363a`:

```
> ssh plutosdr
# iio_info 
Library version: 0.24 (git tag: v0.24)
Compiled with backends: local xml ip usb serial
IIO context created with local backend.
Backend version: 0.24 (git tag: v0.24)
Backend description string: Linux pluto 5.10.0-98231-g9dfba10b795d-dirty #1 SMP PREEMPT Thu Dec 1 13:32:23 SAST 2022 armv7l
IIO context has 8 attributes:
	hw_model: Analog Devices PlutoSDR Rev.C (Z7010-AD9363A)
	hw_model_variant: 1
	hw_serial: 
	fw_version: v0.35-1-g714c-dirty
	ad9361-phy,xo_correction: 40000000
	ad9361-phy,model: ad9363a
	local,kernel: 5.10.0-98231-g9dfba10b795d-dirty
	uri: local:
IIO context has 4 devices:
	iio:device0: ad9361-phy
		11 channels found:
			altvoltage1: TX_LO (output)
			8 channel-specific attributes found:
				attr  0: external value: 0
				attr  1: fastlock_load value: 0
				attr  2: fastlock_recall ERROR: Invalid argument (22)
				attr  3: fastlock_save value: 0 124,124,124,124,124,124,124,124,124,124,124,124,124,124,124,124
				attr  4: fastlock_store value: 0
				attr  5: frequency value: 2450000000
				attr  6: frequency_available value: [325000000 1 3800000000]
				attr  7: powerdown value: 0
```
while based on the comment from @DeonMarais64 it should be different: https://github.com/plutoplus/plutoplus/issues/26#issuecomment-1566127458

U-boot env seems to be correct:
```
# fw_printenv 
adi_loadvals=fdt addr ${fit_load_address} && fdt get value fdt_choosen /configurations/${fit_config}/ fdt && fdt get addr fdtaddr /images/${fdt_choosen} data && fdt addr ${fdtaddr}; fdt get value model / model; if test "${model}" > "Analog Devices Pluto"; then run adi_loadvals_pluto; fi; 
adi_loadvals_pluto=if test -n "${ad936x_ext_refclk}" && test ! -n "${ad936x_skip_ext_refclk}"; then fdt set /clocks/clock@0 clock-frequency ${ad936x_ext_refclk}; fi; if test -n "${ad936x_ext_refclk_override}"; then fdt set /clocks/clock@0 clock-frequency ${ad936x_ext_refclk_override}; fi; if test -n "${refclk_source}" && test ! "${refclk_source}" = "internal" && test ! "${refclk_source}" = "external"; then setenv refclk_source internal; saveenv; fi; if test "${refclk_source}" = "internal" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)" ; then fdt rm /amba/gpio@e000a000/clock_extern_en; fi; if test "${refclk_source}" = "external" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)" ; then fdt rm /amba/gpio@e000a000/clock_internal_en; fi; if test  "${attr_val}" = "ad9361" && test ! "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)" ; then setenv attr_val ad9363a; saveenv; fi; if test -n "${attr_val}" && test ! "${attr_val}" = "ad9361" && test ! "${attr_val}" = "ad9363a" && test ! "${attr_val}" = "ad9364"; then setenv attr_val ad9363a; saveenv; fi; if test -n "${mode}" && test ! "${mode}" = "1r1t" && test ! "${mode}" = "2r2t"; then setenv mode 1r1t; saveenv; fi; if test -n "${attr_name}" && test -n "${attr_val}"; then fdt set /amba/spi@e0006000/ad9361-phy@0 ${attr_name} ${attr_val}; fi; if test "${mode}" = "1r1t" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)"; then fdt rm /amba/spi@e0006000/ad9361-phy@0 adi,2rx-2tx-mode-enable; fdt set /fpga-axi/cf-ad9361-dds-core-lpc@79024000 compatible adi,axi-ad9364-dds-6.00.a; fi; if test -n "${cs_gpio}" && test "${model}" = "Analog Devices PlutoSDR Rev.C (Z7010/AD9363)"; then fdt set /amba/axi_quad_spi@7C430000/ cs-gpios "<0x06 ${cs_gpio} 0>"; fi; if test -n "${attr_val}" && test "${attr_val}" = "ad9364"; then fdt set /fpga-axi/cf-ad9361-dds-core-lpc@79024000 compatible adi,axi-ad9364-dds-6.00.a; if test ! "${mode}" = "1r1t"; then fdt rm /amba/spi@e0006000/ad9361-phy@0 adi,2rx-2tx-mode-enable; setenv mode 1r1t; saveenv; fi; fi; 
baudrate=115200
bitstream_image=system.bit.bin
bitstream_size=0x400000
bootcmd=run $modeboot
bootdelay=3
bootenv=uEnv.txt
boot_image=BOOT.bin
boot_size=0xF00000
clear_reset_cause=mw f8000008 df0d && mw f8000258 00400000 && mw f8000004 767b
devicetree_image=devicetree.dtb
devicetree_load_address=0x2000000
devicetree_size=0x20000
dfu_mmc_info=set dfu_alt_info ${kernel_image} fat 0 1\\;${devicetree_image} fat 0 1\\;${ramdisk_image} fat 0 1
dfu_mmc=run dfu_mmc_info && dfu 0 mmc 0
dfu_ram=echo Entering DFU RAM mode ... && run dfu_ram_info && dfu 0 ram 0
dfu_ram_info=set dfu_alt_info dummy.dfu ram 0 0\\;firmware.dfu ram ${fit_load_address} 0x1E00000
dfu_sf=gpio set 15;set stdout serial@e0001000;echo Entering DFU SF mode ... && run dfu_sf_info && dfu 0 sf 0:0:40000000:0 && if test -n ${dfu_alt_num} && test ${dfu_alt_num} = 1; then set fit_size ${filesize} && set dfu_alt_num && env save; fi;gpio clear 15;
dfu_sf_info=set dfu_alt_info boot.dfu raw 0x0 0x100000\\;firmware.dfu raw 0x200000 0x1E00000\\;uboot-extra-env.dfu raw 0xFF000 0x1000\\;uboot-env.dfu raw 0x100000 0x20000\\;spare.dfu raw 0x120000 0xE0000
ethaddr=00:0a:35:00:01:22
extraenv_load_address=0x207E000
fdt_high=0x20000000
fit_config=config@0
fit_load_address=0x2080000
fit_size=0x900000
importbootenv=echo Importing environment from SD ...; env import -t ${loadbootenv_addr} $filesize
initrd_high=0x20000000
ipaddr=192.168.2.1
[... 47 lines cut ...]
and some variations from https://www.rozenek.com/adalm-pluto-rev-c-hacks-how-to-extend-frequency-range-add-gps-do-and-add-rx2-and-tx2/
```
fw_setenv attr_name compatible
fw_setenv attr_val ad9364
fw_setenv maxcpus
pluto_reboot reset
```

With the following firmwares:
1. [Official v0.38 from AD](https://github.com/analogdevicesinc/plutosdr-fw/releases/tag/v0.38)
2. [v0.38 from @azolotko](https://github.com/azolotko/plutoplus/releases/tag/v0.38)
3. [v0.37 with Ethernet from @DeonMarais64](https://github.com/DeonMarais64/PlutoPlusSDR-FW/releases/tag/v0.37%2B)
2. [v0.35 with Ethernet from @DeonMarais64](https://github.com/DeonMarais64/PlutoPlusSDR-FW/releases/tag/v0.35%2B)
4. [v0.33 original](https://github.com/plutoplus/plutoplus/releases/tag/v1.33)
5. A few others, v0.34 or v.032 found in comments in other issues.

I tried flashing both DFU and mass storage/FRM.

### Device
Pluto+ v2 (with 3.3v pin)

### Observations
It seems the unit just ignores everything in U-Boot env which is accessible via `fw_setenv` or `fw_printenv`. When I flash the device via DFU (`firmware.dfu`, `boot.dfu`, `uboot-env.dfu`) with non-AD firmwares and try to `fw_printenv`, I got errors like `Warning: Bad CRC, using default environment` and just a few lines output (just baudrate and bootcmd). The device works though. If I try to set any env variable (or write everything with `fw_setenv -s file_with_all_env`) then it bricks the device - it seems that this command overrides something. When I reflash it via DFU - the device back to life, and magically the env is set properly then. If I change something there, it's just ignored during the boot. For example, with this method I also always had only one cpu available, despite  `fw_setenv maxcpus 2` set - it seems only default config was used all the time.

Flashing via FRM (boot.frm+firmware.frm or putting the archive itself into mass storage) seems more stable, I'm able to set `fw_setenv` or `fw_printenv` without issues after that. And I also noticed that now I had 2 cpus. However changing this value with `fw_setenv maxcpus 1` doesn't work and I still has 2 cpus. This makes me think that there is an issue with U-Boot env or where it is flashed, but I'm not sure what to do next with this.

jmsigler, 2024-07-27:
Hi belashou seems we have the same issue. I'm also seeing weird behavior with the uboot env.

I soldered a header to the board to view the uboot boot log and the interesting thing is that the env variables seem fine there and it isn't until later in the boot process that something goes wrong w/ the CRC. Unfortunately the log isn't very verbose so I'm not really sure where it is failing there. 

I see the same behavior when I directly change an env variable from uboot. I see that it is persistent across boots, however, the boot doesn't actually use the value. I tried changing the eth addr for the default interface and it had no affect in linux for example.

I have a feeling it is related to the official firmware version 0.38. I also flashed this fw and then started having these issues after I went back to the pluto sdr plus fw. Possibly it did something with jffs2 mount or something that is messing things up.

[uboot_log.txt](https://github.com/user-attachments/files/16397477/uboot_log.txt)

DeonMarais64, 2024-07-27:
I have not tried v0.38. However, I am wondering the problem may have something to do with changes ADI made for Phaser support in v0.38?

jmsigler, 2024-07-28:
I did an experiment where I completely erased the flash via uboot 

`sf erase 0x0 0x2000000`

I then re-dfued the firmware (using Deon's 0.37 to start with and also tried 0.35) with no success (Still getting the BAD ENV errors). At this point wondering if it is a hw bug because the flash is the only persistent storage on the device so erasing it should be a fresh start. Only other thing would be a fuse or something but I don't think thats the case really.

jmsigler, 2024-07-28:
@DeonMarais64 Have your builds been tested on Pluto Plus V2 harware by chance? Just curious since HW differences are the only thing I can think of that would be causing this now.

jmsigler, 2024-07-28:
I reverted to 0.32 from here: https://github.com/plutoplus/plutoplus/issues/10#issue-986244383 and that is working for me with respect to the env variables, so not a hardware issue. Definitely something that was changed in the software.

jmsigler, 2024-07-29:
Ok more progress, at this point I think it is an issue with the commands in the uboot-env its self. Maybe something in there isn't compatible with Pluto Plus V2 or there is some other issue.

Starting with Pluto Plus 0.32 on my device I can dfu 0.38 from Azolotko and env vars are fine (i.e. no CRC errors) and match the 0.32 as expected. However, when I take the next step of DFUing Azolotko's uboot-env.dfu then this breaks things.

azolotko, 2024-07-29:
> However, when I take the next step of DFUing Azolotko's uboot-env.dfu then this breaks things.

Thanks @jmsigler! I’ll take a look.

jmsigler, 2024-07-29:
Ok so I think I figured it out after a bit of digging.

As part of the boot process uboot runs the command "read_sf" which loads data from flash into memory for boot. For some reason when read_sf is run for the first time it fails. Not sure why, maybe our particular revision of Rev 2 has some problem with the power sequence or loading 0.38 damaged it or something else. My workaround for this was to modify the bootcmd so that it does an initial read_sf so that the second one that is part of the actual boot works correctly. This seems to have fixed my issues. The other puzzle is why it was working with 0.32... As far as I can tell 0.32 doesn't really have any differences w/ its read_sf or qspiboot variables so it should have been behaving the same.

To do this go into uboot (you will need to solder a header to the TX/RX pins on the connector above J4). 

`setenv bootcmd 'run read_sf; run $modeboot'`

If you're still having an issue with your frequency extension you may also need to set the model, fit_config, and fdt_choosen variables as well. Your specific issue might need more investigation but this at least seems to fix the uboot env bug we're both having.

`setenv model 'Analog Devices PlutoSDR Rev.C (Z7010/AD9363)'`
`setenv fit_config 'config@9'`
`setenv fdt_choosen 'fdt@3'`

DeonMarais64, 2024-07-29:
> @DeonMarais64 Have your builds been tested on Pluto Plus V2 harware by chance? Just curious since HW differences are the only thing I can think of that would be causing this now.

I only have Pluto Plus V2 hardware and my builds all work on them (I have two of them). All official ADI firmware prior to v0.38 work (I have not yet tried V0.38).  However, there are some limitations when using the stock ADI firmware after v0.33. On versions after v0.33, the hack to ad9361 for extended frequency range or tx2rx2 does not work because ADI only allow that on RevC  of the standard Pluto (and not on RevB) and the Pluto Plus is closer to RevB. This is reason for one of the patches for the Pluto Plus builds,

```
diff --git a/scripts/pluto.its b/scripts/pluto.its
index e69e417..aa7fa08 100644
--- a/scripts/pluto.its
+++ b/scripts/pluto.its
@@ -156,8 +156,8 @@
 		};
 
 		config@9 { /* This one is actually RevB */
-			description = "Linux with fpga RevB";
-			fdt = "fdt@2";
+			description = "Linux with fpga RevC";
+			fdt = "fdt@3";
 			kernel = "linux_kernel@1";
 			ramdisk = "ramdisk@1";
 			fpga = "fpga@1";
```

jmsigler, 2024-07-29:
Ah maybe its not correct then to set setenv model 'Analog Devices PlutoSDR Rev.C (Z7010/AD9363)'.  Do you agree that setenv fit_config 'config@9' and setenv fdt_choosen 'fdt@3' are correct for the Pluto Plus?

Since @belashou and I both had this problem around the same time maybe it is an issue with the run of pluto pluses we happen to buy. Very strange bug

DeonMarais64, 2024-07-29:
If my understanding is correct there are now at least two instances of irreversible issues which seem to have been caused by loading of v0.38. And going back to an earlier version that worked correctly prior to having loaded v0.38 does not correctly re-instantiate the expected desired behavior. This seems to me (at least feasibly) that FW v0.38 expects a RevC hardware layout and programs some of the IO in a fashion that will cause irreversible damage on RevB hardware. The Pluto Plus patch above is spoofing RevC hardware on what is effectively RevB hardware, so this is at least a possibility and worth looking at as an explanation.

Looking at v0.38 release notes, https://github.com/analogdevicesinc/plutosdr-fw/releases/tag/v0.38 
what jumps out at me is the addition of phaser support which involves GPIO. The details can be found in the associated links.  
This was why I made the suggestion above.

At some point I looked at the differences between the ADI hardware versions and that of the Pluto Plus (as lousy as the documentation is), I will try to dig out and post what I summarized.

belashou, 2024-07-30:
@jmsigler @DeonMarais64 appreciate your comments! Two cents from my side:

1. When it comes to the original issue (AD9361 hack and frequency unlock) I managed to partially solve this problem when recompiled @azolotko's v0.38 version with env variables explicitly set in [zynq-common.h](https://github.com/analogdevicesinc/u-boot-xlnx/blob/a2c2013a86231af6a61d6e7dca6c1e28a3bc57bf/include/configs/zynq-common.h#L235). After that, my Pluto plus recognized ADC as AD9361 and allowed me to catch some FM station. I will push source/binaries this week. At least there is no hardware issue with this. I guess Ethernet address can be also changed there for @jmsigler, however this workaround is pretty clumsy.
1.1 Obviously, issues with u-boot env and CRC errors are still there.
1.2 I noticed an interesting fact while checking [update.sh](https://github.com/analogdevicesinc/buildroot/blob/f70f4aff40bcc16e3d9a920984d034ad108f4993/board/pluto/update.sh#L166) script. u-boot env is accessible via `/dev/mtdblock1` (or /dev/mtd0). When I'm reading this sector (or any other like `/dev/mtd0` or `/dev/mtd3`) with `dd if=/dev/mtdblock1 bs=64k ` then I get complete garbage - nothing even closely similar to u-boot/firmware/u-boot env binary. If I override u-boot env sector `dd if=/opt/u-boot-env.bin of=/dev/mtdblock1 bs=64k` then I obviously can read env variables with `fw_printenv` without issues until reboot - after reboot the unit is bricked, and firmware sector requires recovery. At this point, I'm about to blame MTD driver, or partition layout.
2. I'm a bit hesitant to blame hardware as for now simply because there is only one flash on the board, which contains u-boot, env, spare sector and firmware. Any hardware issues will probably impact u-boot or firmware itself, but so far it works. To my understanding, the issue appears on the higher abstraction level. However, I do not exclude some low level issues caused u-boot/MTD being crazy.
3. I was checking [Phaser commit as well](https://github.com/analogdevicesinc/hdl/commit/763ee38ee00f2b625824643f1f3f32c107f47444) as needed to correct pins for v.038, but it seems it impacts mostly MISO/MOSI/SDA/SCL/CLK which leads to pin header only (based on schematics). I would carefully assume that Phaser integration shouldn't be the root cause. The fact that we had issues on previous firmware versions contributes to this fact (well, until there is no any permanent damage done to the board with all our experiments as @DeonMarais64 mentioned above :laughing: ).
4. I like the dive deep from @jmsigler, I soldered headers as well and try to enter the u-boot console this week to play with device tree and will try to track down differences. Double `read_sf` sounds crazily genially, and definitely worth digging deeper.

Also, at this point, I'm wondering if it was any firmware version that had no CRC errors for u-boot env AND where env vars were actually used (i.e. it was possible to change Ethernet address, hack AD9361, change CPU count or anything else)? Apologies if I missed something. If it is, I will try to flash it, confirm that it works, and then try to compare sources.

stargo, 2024-08-05:
I also had the CRC errors after flashing newer firmware. I tried a few different things and then went back to the stock firmware v0.32-1-g7bdc-dirty which I had saved and the CRC errors were gone. I couldn't find that firmware anywhere to download, so I've uploaded it here: https://rmdir.de/~michael/plutoplus/plutoplus-v0.32-1-g7bdc-dirty.tar.xz

I've nulled out the spare.dfu as it contains stuff like ssh host keys and gets regenerated on boot (just don't flash it) and the uboot-env contains my ethernet config, so it's not completely as from the factory.

lucabicego, 2024-10-05:
Hi, i'm sorry but i'm a newbie on this topic. I've bought a Pluto + and, like suggested, i upgraded to v 038 of the firmware. Now i've the same issue ie i cannot extend range to 70-6000 Mhz. What i can do now ? i try to revert the firmware to v 037 but it doesn't work. Could you please tell me what i have to do to solve this problem. Thanks.

SchattenWolf2008, 2024-11-12:
Hello

I was having the same issues.

I was able to use it fine with the plutoplus 37 dirty version, but I was not able to make any uboot env changes.

I downgraded back to v0.32 that @stargo provided, and it works but now I am back to the same issue which was also the reason why I tried to upgrade in the first place:

For some reason, the plutoplus sdr seems to connect with the network via the dedicated Ethernet port perfectly fine but cancels connections after three minutes. After that I cannot access SSH, the Website or the RTL interface.

This is quite frustrating to me, as I am neither able to use the full 70-6000MHz range due to the env issues with the newer firmware, and I am unable to use the network connection on the v0.32 firmware.

As this post has been open for quite a long time now I wanted to ask if someone has found a solution for this?

Could you kindly share?

I would really appreciate it.

Just to reconfirm, I have the Pluto+ SDR Revision 2 (3.3v)

SchattenWolf2008, 2024-11-12:
Well I have installed v0.32 again as it seems to be the only version where I can make changes.
Its the only version where I seem to get the full 70 - 6000MHz range, which is working great.

The only issue is, that it simply wont connect using the ethernet plug.

I can use it with USB only and its quite frustrating as this is the only version that works flawlessly, except this network issue.

It seems to get the ip address and get registered in the router, however, any connections via ethernet dont work.

Neither SSH, the Website or RTL TCP.

Did anyone find a solution for this problem? USB simply has too less bandwith and all the other versions dont work properly. If someone knows how I can get the ethernet working, please let me know.

config.txt:

`# Analog Devices PlutoSDR Rev.C (Z7010-AD9363)
# Device Configuration File
#   1. Open with an Editor
#   2. Edit this file
#   3. Save this file on the device USB drive
#   4. Eject the device USB Drive
# Doc: https://wiki.analog.com/university/tools/pluto/users/customizing

[NETWORK]
hostname = pluto
ipaddr = 192.168.1.106
ipaddr_host = 192.168.1.1
netmask = 255.255.255.0

[WLAN]
ssid_wlan = 
pwd_wlan = 
ipaddr_wlan = 

[USB_ETHERNET]
ipaddr_eth = 
netmask_eth = 255.255.255.0

[SYSTEM]
xo_correction = 
udc_handle_suspend = 0

[ACTIONS]
diagnostic_report = 0
dfu = 0
reset = 0
calibrate = 0
`

Everything else is unmodified.

SchattenWolf2008, 2024-11-12:
Hello everyone, I have found a solution which seems to work.

For the firmware I am using F5OEO/plutosdr-fw (https://github.com/F5OEO/plutosdr-fw/releases/download/PlutoDVB2_0.5.16.2/plutoplus-fw-0.5.16.2.zip)

And I have figured out, for some reason, I need to put the IP for the physical ethernet port at [USB_Ethernet] and the IP for the USB connection at [NETWORK].

Now its working for me :)

mathijsvandenberg, 2024-11-12:
Just out of curiosity. Does this also work like with SDR++ being able to connect by network instead of USB?

SchattenWolf2008, 2024-11-12:
> Just out of curiosity. Does this also work like with SDR++ being able to connect by network instead of USB?

Yup
[image]

F5OEO, 2024-11-12:
Beta, but  f you want any try with sdr++/plutoplus high bandwidth (34Mhz) ,extended frequencies 47Mhz-6Ghz try https://github.com/F5OEO/tezuka_fw/releases/download/0.0.5/plutoplus_maiasdr.zip with MIO46 jumper.
[image]

lucabicego, 2024-11-13:
Hi i used the same firmware and it works even for me !

I’m using my PlutoPlus with SDR++ with an USB connection.

Thanks a lot to notify me this firmware !

Best regards.

Luca Bicego