plutoplus/plutoplus #50: Tezuka ; New firmware and builder for plutoplus (and others)

F5OEO, 2024-11-10:
Project is still in beta but could be usefull to update with new features. (SD card, Frequency extension, Complex 8bit for higher bandwidth...)
See at https://github.com/F5OEO/tezuka_fw
Already some releases

Thanks for your feedback.

SchattenWolf2008, 2024-11-12:
Installed this and my Pluto+ SDR is now completely bricked. Thanks.

mathijsvandenberg, 2024-11-12:
I've installed it at works like expected. However I had to remove the jumper bridge from USRT- MIO52.

F5OEO, 2024-11-12:
I assume the Jumper is on USRT-MIO_46. I tested on 1.8 and 3.3 plutoplus version. USRT-MIO52 has maybe to be fixed (sorry). The main issue with plutoplus is the SPI flash not protected..thus could be easily corrupted by an update. Maybe prefere to boot on SD card (sdimg folder content has to be copied to a fat32 sd card and jumper SD-H to 1V8).
Thanks for feedback, and sorry for inconvenience...

F5OEO, 2024-11-12:
Hope that you still have u-boot. Have you put a serial debug to see what's happening?
 I have successfully flash a brick plutoplus with jtag and openocd. I begin to try with dirtyjtag on pico, but not success tight now. I think it could be a very usefull way of unbricking.
Let me know if you have any question.

SchattenWolf2008, 2024-11-12:
No worries, I unbricked it using two Raspberry Pi Picos and a few jumper cables.

I am currently flashing the original firmware and will now try yours with the different jumper setting

SchattenWolf2008, 2024-11-12:
So I tried installing it again, making sure the jumper is on URST-MIO46.

When it finishes flashing and restarts, the green LED shows up for a few seconds and then goes off.

No signs of anything.

SchattenWolf2008, 2024-11-12:
> Nice for jtag with pico...curious about procedure you use ? Dirtyjtag ? Use jumper on URST-MIO46

Haha it works more on hopes and prayers.

I am using two picos:
Pico 1: JTAG connector
- using xvcd-pico
- connecting the pins accordingly to the documentation
- downloading the latest official firmware jtag-bootstrap file
- connecting and selecting the Pluto+ SDR beforehand in Vivado
- running command "exec xsdb run-xsdb.tcl" while pressing / holding the DFU button, jumper set on URST-MIO52

Pico 2: UARST connector
- using pico-uart-bridge-v4.2 (github release)
- just flash it onto the pico, connect the UART TX, RX and GND pins accordingly
- Connect via Serial on PuTTY 115200 BAUD (Serial 1)
(tho I dont see anything happening in there, I rather just have it running and assume it somehow helps with the process)

Then after a bit of patience it should work

SchattenWolf2008, 2024-11-12:
> I assume the Jumper is on USRT-MIO_46. I tested on 1.8 and 3.3 plutoplus version. USRT-MIO52 has maybe to be fixed (sorry). The main issue with plutoplus is the SPI flash not protected..thus could [...]

I tried with the SD card but both green and blue LED just show up constantly, nothing else happens.
Quite unfortunate as I am really interested into this firmware.

I have a question, did you maybe build it from v0.33? Thats the only firmware which causes this big bricking issues

SchattenWolf2008, 2024-11-12:
Good news: When I completely reflash boot.dfu and pluto.dfu your version boots up.

However, you havent uploaded the uboot env file, right? I cannot make changes due to invalid uboot enviroment error.

SchattenWolf2008, 2024-11-12:
I have managed to install your firmware by flashing boot.dfu and pluto.dfu from DFU mode, I was able to change the settings of config.txt successfully, however, the Ethernet connection does still not work.

It does connect to the router and get its IP, but it does not allow any connections to SSH HTTP or RTL_TCP

Thats pretty much the same issue I had on the v0.32 firmware.

With the newest official firmware I think I did not have any issues with the ethernet connection, but well, that firmware is limited to 325 MHz - 3 GHz and I was unable to change settings there.

Can you maybe help me with the ethernet issue?

F5OEO, 2024-11-12:
Good news about updating the firmware...I propose to contact me directly evaristec@gmail.com to solve the last issues.

SchattenWolf2008, 2024-11-12:
> Good news about updating the firmware...I propose to contact me directly [evaristec@gmail.com](mailto:evaristec@gmail.com) to solve the last issues.

Sure!

Do you have Discord?

I emailed you.

SchattenWolf2008, 2024-11-12:
Hey I got it working

Thanks :)

CryptoIceMLH, 2024-11-16:
i am also interested in getting this working. I have a plutoplus and only got official 0.39 working on it.. a bit of a noob with linux so trying to figure out an easy way to get the Pluto plus to work over ethernet. I am building a QO100 tx / rx

SchattenWolf2008, 2024-11-16:
> i am also interested in getting this working. I have a plutoplus and only got official 0.39 working on it.. a bit of a noob with linux so trying to figure out an easy way to get the Pluto plus to work over ethernet. I am building a QO100 tx / rx

Hey, maybe check out my post 
#51 scroll down to #FAQ Found a working Pluto+ SDR Rev 2 firmware with working Ethernet and config.txt (#firmware #ethernet #uboot)