plutoplus/plutoplus #51: The Solution for most our problems (FAQ Attempt)

SchattenWolf2008, 2024-11-13:
Hello, this post aims to be more like a "FAQ Space" for most our problems to find solutions.
If you have anything, you can add as a comment, ofc, you can also ask but try to put #FAQ or something at the beginning and some texts that can be used to find easier via the search feature (CTRL + F)

SchattenWolf2008, 2024-11-13:
#FAQ Found a working Pluto+ SDR Rev 2 firmware with working Ethernet and config.txt (#firmware #ethernet #uboot)

**Pre-story:**
It is very tricky to find working Firmware for the Pluto+ SDR, as you'll mostly only find Firmware for the Adalm Pluto, or many firmwares dont work on the newer Rev 2 boards.

Especially Firmware such as v0.33, which worked fine on the Pluto+ SDR Rev 1 do not work on the Rev 2 boards anymore, and v0.33 actually bricks the device.
(If your PlutoSDR shows 3.3V near the JTAG connections (left of Jumper) you have a Rev 2 board; Rev1 will show 1_8V)

I personally had the issue, that firmware that didnt work well (e.g. not full frequency range, unable to change config.txt) work fine with the Ethernet. However, I was unable to change any settings or patch it to support the full 70 - 6000MHz range.

Other firmware supported the full frequency range, allowed me to edit the config.txt or allowed me to patch it, but while it was connecting with Ethernet I was just not able to access SSH, Web or the radio interface via Ethernet, while it was working fine with the USB connection.

After a lot of digging, I have finally found what seems to be a fairly up to date, working firmware that was supporting my Pluto+ SDR natively (full frequency range), had working ethernet and didnt have a lot of other stuff installed I didnt use.

**Setup:**
For setting it up, I usually prefer putting the device in DFU (Jumper URST-MIO52, while plugging in hold DFU)

Then I use dfu-util for installing the firmware, simply [download ](https://dfu-util.sourceforge.net/releases/)and extract it, open CMD and run dfu-util from there.
(Lastest release on the bottom, download dfu-util-0.11-binaries.tar.xz, extract folder win64 somewhere)

Open CMD

Navigate to where you extracted dfu-util/win64

Lets say if we extracted it to D:\data\downloads\dfu-utilwin64\ it would be

`cd D:`
`cd data`
`cd downloads`
`cd dfu-utilwin64`

Now [download and extract the firmware files](https://github.com/F5OEO/plutosdr-fw/releases/download/PlutoDVB2_0.5.16.2/plutoplus-fw-0.5.16.2.zip) that we will be using, into the directory of where you extracted dfu-utils to. 

We will need the files boot.dfu and plutoplus.dfu (which is usually called pluto.dfu), and the ubuut-env.dfu

Now in the CMD window we will type the following commands for installing the firmware:

`dfu-util -a firmware.dfu -D ./plutoplus.dfu`
`dfu-util -a boot.dfu -D ./boot.dfu`
`dfu-util -a uboot-env.dfu -D ./uboot-env.dfu`

After this is done, you can unplug the Pluto+ SDR, set the jumper back to URST-MIO46 and plug the Pluto+ SDR back in.

It should now boot up with the new firmware.

**Setting up Ethernet:**

It is quite funny, but in my case atleast I had to put the IP address for the physical Ethernet connector in the [USB_Ethernet] section, and the USB connection is configured by [NETWORK]

So lets assume we connect the Pluto+ SDR to a router and give it the IP address 192.168.1.106, while assigning it the IP address 192.168.2.1 for the USB connection, our config will look like this:

```
(This is the config.txt from a different firmware, but the idea is the same)
# Analog Devices PlutoSDR Rev.C (Z7010-AD9363)
# Device Configuration File
#   1. Open with an Editor
#   2. Edit this file
#   3. Save this file on the device USB drive
#   4. Eject the device USB Drive
# Doc: https://wiki.analog.com/university/tools/pluto/users/customizing

[NETWORK]
hostname = pluto
ipaddr = 192.168.2.1
ipaddr_host = 192.168.2.2
netmask = 255.255.255.0

[WLAN]
ssid_wlan = 
pwd_wlan = 
[... 78 lines cut ...]
[pluto:~]# ifconfig -a
eth0      Link encap:Ethernet  HWaddr 00:00:00:00:00:00
          inet addr:192.168.1.106  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1171 errors:0 dropped:153 overruns:0 frame:0
          TX packets:221 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:132027 (128.9 KiB)  TX bytes:18797 (18.3 KiB)
          Interrupt:35 Base address:0xb000

... (other interfaces)

usb0      Link encap:Ethernet  HWaddr 00:00:00:00:00:00
          inet addr:192.168.2.1  Bcast:0.0.0.0  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:642749 errors:0 dropped:262 overruns:0 frame:0
          TX packets:1281575 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:25829844 (24.6 MiB)  TX bytes:1987154289 (1.8 GiB)
```
          
And it allows me to connect as expected.

I'm not sure wether this issue is only for me or not, but it works now.
I assume I had the same issue with the original v0.32 too, but I'm not sure.

F5OEO, 2024-11-14:
On both revisions of pluto+ , levels are 1.8 Volt. I am surprised that you plugin directly 3.3v level pin of pico without level convertor  for jtag. Can you confirm ?

SchattenWolf2008, 2024-11-14:
> On both revisions of pluto+ , levels are 1.8 Volt. I am surprised that you plugin directly 3.3v level pin of pico without level convertor for jtag. Can you confirm ?

On Rev. 2 Boards it uses 3.3V, however, on boards that use 1_8V you have to use some sort of resistor to reduce the interfacing voltage.

videobuff, 2024-12-27:
Good afternoon and i hope you can point me in the right direction.
I followed your procedure and all went ok until I came to the chapter where you have to:
`Now in Vivado you can open the hardware manager, click on auto connect, and right click on that --> Add Xilinx Virtual Cable (XV`C)

There is a timeout opening port to xvc server localhost 2542

Situation:
Pico: connected to the Pluto + (rev2 with 3.3V) 
I connected all signals as well as the ground to the Pluto +

Pluto connected to win11 pc with its own USB cable
Changed the 
Pico also connected to the same Win11 pc. Flashed it succesfully with xvcPico.uf2. (sound is heard and i see the drectory)

Vivavo started and via autoconnect finds localhost(0), but refuses the virtual cable command due to timeout.
The jumper is over MO52 and URST

Thanks for any help already....

Erik PA0ESH

m0boy, 2025-01-03:
Wow, I found a fix for this without having to use any JTAG.
So, I bricked my brand new Pluto SDR + V2 (3V3 levels) with a firmware that as said above was not suitable.

Going back and forth with what is said above and studying the JTAG option, I found @F5OEO  statement saying: 

>  The main issue with plutoplus is the SPI flash not protected..thus could be easily corrupted by an update. Maybe prefere to boot on SD card (sdimg folder content has to be copied to a fat32 sd card and jumper SD-H to 1V8).

1. Based on the above, I shorted SD_H to 3V3 (V2 board remember? )
2. Left the jumper on MIO46
3. copied the sdimg folder content of the firmware above in @SchattenWolf2008 's post onto a fresh SD card.
4. inserted the card into the pluto plus
5. plugged in a usb into the pluto plus with the SD card in
Surprise surprise, the pluto sdr boots! Works! 

Allright, you would say, of course, because it is booting from the SD card.

6. Now go onto the removable disk drive the pluto sdr creates, that partition where you have config.txt.
7. Copy the pluto.frm and boot.frm from same firmware you took the sdimg stuff from, into the root folder of that removable drive. 
8. Eject the drive, wait until the fast blinking stops plus at least another 10seconds. (took about 3 minutes to complete)
9. Disconnect plutosdr plus completely from usb and/or power.
10. remove SD_H jumper from 3V3 line, leave jumper on MIO_46
11. connect pluto to USB, wait 10 seconds , blue will start blinking!
12.  ENJOY!

Lowsp00f, 2025-03-26:
The guys helped a lot, I already soldered the comb for jtag and ordered another pluto.As a result, I had a problem with the 8gb card. I tried to boot from it for a week, after writing boot to another sector, Finally resigned, I poked the 128gb card and lo and behold!

wolfgang6444, 2025-07-16:
I managed to unbrick and can connect via USB (minicom) - no SD-card.
Now I am struggeling with the network setup:

I want to assign a fixed IP say 192.168.2.68 to the interface on the RJ45.
I do not want to use the USB-IP.
I cannot use 192.168.2.1 - that's my router.

ifconfig -a
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:14 errors:0 dropped:0 overruns:0 frame:0
          TX packets:14 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1585 (1.5 KiB)  TX bytes:1585 (1.5 KiB)

usb0      Link encap:Ethernet  HWaddr 00:05:F7:19:E7:93  
          inet addr:192.168.2.1  Bcast:0.0.0.0  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:108 errors:0 dropped:13 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:21932 (21.4 KiB)  TX bytes:1787 (1.7 KiB)

Whenever I edit config.txt on the massstorage it gets overwritten with the next reboot - yes I umount it to store.
There is a config.txt at /opt/config.txt on the filesystem of the pluto+  Is this what I have to edit?
Is there an editor on the pluto+

Wolfgang

wolfgang6444, 2025-08-03:
#question
I am using the recent tezuka-fw from F5OEO. I am trying to operate the plutoplus in 2r2t mode. But in gnuradio and also with the examples from ADI I always get the error message, that the second channel is not available on the block.
I set the fw_switches according to  https://wiki.analog.com/university/tools/pluto/users/customizing 👍 

# fw_setenv attr_name compatible
# fw_setenv attr_val ad9361
# fw_setenv compatible ad9361
# fw_setenv mode 2r2t
# reboot

Does somebody know what I need to do to enable the second RX and TX?

Many thanks

Wolfgang

antonpogrebenko-public, 2025-08-03:
You can use the plutoplus firmware instead. It has 2r2t mode enabled out of
the box

antonpogrebenko-public, 2025-08-03:
> Do you mean plutoplus from F5OEO or another source?

https://github.com/plutoplus/plutoplus/pull/36#issuecomment-2853546453

wolfgang6444, 2025-08-03:
I see, but I want to use my plutoplus remotely, not attached to any PC via USB so I am relying on the Gigabit RJ45

antonpogrebenko-public, 2025-08-03:
When I power on the Pluto, it stays undiscoverable till the moment I log in
and use the eth0 interface (ex: ping command). Then it can be disconnected
from the computer usb and used remotely over the RJ. I agree, that’s a
workaround, but I didn’t manage to find another solution.

wolfgang6444, 2025-08-03:
Hm, does not work for me:

I can connect and login to the plutoplus via minicom --device=/dev/ttyACM0
Then I can ping my pluto from the host it is connected to at address 192.168.2.1 BUT ONLY IF I DISCONNECT THIS HOST FROM THE REST OF MY NETWORK - otherwise 192.168.2.1 goes to my router and not the pluto.

This is a very weird concept to hardwire the IP of the pluto to 192.168.2.1

iz8huj-ui, 2025-10-17:
#FAQ 
Hi guys, I got a pluto+, and after upgrading @F5OEO's firmware, the pluto+ still has a solid green LED, and I can no longer access it via USB.
I tried charging via USB, using the jumper and pressing the DFU button, as @M0BOY described, but I don't notice anything different. The green LED is solid. It still doesn't respond. Do you have any other suggestions?
Thanks

videobuff, 2025-10-17:
Try this : https://www.m0boy.radio/how-to-unbrick-a-pluto-sdr-without-jtag/

arsatyants, 2026-01-15:
Ive got absolutely the same situation. Was you succeeded to get back it live?