plutoplus/plutoplus #53: What kind of Board / Version of Pluto+ SDR is that ?

helmarw, 2025-03-23:
Hi, i just got a Pluto+ SDR from Banggood (https://www.banggood.com/Pluto+-70MHz~6GHZ-AD9363-SDR-Radio-Receiver-Transmitter-2xRX-2xTX-Compatible-with-ADI-ADALM-PLUTO-PLUTO+-p-2027918.html?cur_warehouse=CN&ID=6287830)

After testing a few things successfully out of the box with gqrx and satsagen i was wondering if i could update the firmware.
Just to be sure what kind of board ive got i opened the casing... this came i bit of a surprise .. the board looks nothing
like that in the pictures:

[image]

[image]

There is a switch for the firmware bottom right and a second switch next to the ZYNQ which function left of the ZYNQ i assume.
The Boot/DFU switch S1 is on the bottom of the board, at least i assume thats its function.

Does anybody have this board and more information about it ?

Is this repository still active ?

thnx
Helmar

helmarw, 2025-04-05:
nope, i did not sorry

TechMindsYT, 2025-04-05:
I spoke with designer of original Pluto+, he said that this its a knockoff/clone.  As there is no cover over the components, its possible quality is not good.

One question, did you manage to get this reading TF card from boot up? I could not..

I think I will ditch this board and use original Pluto+ board. At least until more information about this board comes to light.

Thanks
Matt

helmarw, 2025-04-05:
no i havnt tried the SD card yet, but in general its working
i have tezuka_fw 0.1.1 on it

where did you buy the original Pluto+ board ?

formtapez, 2025-04-15:
Where can i buy the "original" version from this repository?

dineshb-ucsd, 2025-05-12:
Where can i buy the "original" version from this repository?

videobuff, 2025-05-13:
Ik heb dit zojuist gevonden op AliExpress: €500,85 | HAMGEEK Originele PLUTO + ADC/DAC AD9363 SDR Radio-ontvanger Zender 70 MHz-6 GHz Software Defined Radio ADI ADALM-PLUTO Plushttps://a.aliexpress.com/_EJc0bpyMet vriendelijke groetenErik Schott

I got mine on Ebay

videobuff, 2025-05-13:
I copied the advert but the price is wrong unless it includes  US tariffs 😂😂I paid 160 euro for it and no tariffs 73’s Erik l

FlorinMatei1957, 2025-05-14:
A few days ago, I also bought the exact same board from: https://www.aliexpress.com/item/1005003636071261.html?spm=a2g0o.order_list.order_list_main.17.6ca81802j2t0ku . After connecting it to the laptop, I upgraded its firmware to ver. 0.39. After numerous attempts, I was unable to connect Pluto+ via ethernet directly to the laptop. It only works via USB, and the reception and transmission tests done so far are satisfactory.

[image]

dineshb-ucsd, 2025-05-14:
Has anyone attempted to order this? 
https://evalm.com/products/48223296?_pos=1&_psq=pluto%2B&_ss=e&_v=1.0

videobuff, 2025-05-14:
Florin. You did adapt the config file in the root with the reverse values ? USB is network and network is USB. Met vriendelijke groetenErik Schott

https://evalm.com/products/48223296?_pos=1&_psq=pluto%2B&_ss=e&_v=1.0

antonpogrebenko-public, 2025-05-14:
> A few days ago, I also bought the exact same board from: https://www.aliexpress.com/item/1005003636071261.html?spm=a2g0o.order_list.order_list_main.17.6ca81802j2t0ku . After connecting it to the [...]

If you are using the official Pluto SDR firmware it doesn’t include ethernet drivers in it.

videobuff, 2025-05-14:
Florin. I have the same firmware but if you connect directly to the laptop you have to set the network adapter in the laptop manually to the same range as the Pluto. So 192.168.2.13 for instance. However, I have my Pluto connected to the home network and my laptop as wel. DHCP gives ip addresses to both so make sure you make an ip  reservation for the Pluto address based on the mac in your router. Then the search function in SDR console should find it. Met vriendelijke groetenErik Schott

IMG20250502130812.jpg (view on web)

FlorinMatei1957, 2025-05-15:
Thank you very much Erik. I will study carefully what you suggested and do some tests. All the best!

SteffenLav, 2025-06-23:
I also got the (wrong/cloned/knockoff) Pluto+ board. However it is working perfectly and reliably so far as intended.
I would like to use my programmable OCXO on it and connected the coax on the marked socket. But where is the ExtClk pin that is supposed to be grounded?? Cannot find it anywhere....

SteffenLav, 2025-07-15:
@TechMindsYT @videobuff @helmarw 

Okay found the missing link on the knock-off board:

It is the MIO48 that needs to go HIGH (1,8V) to switch to the external clock instead of the internal 40MHz (Y3)
MIO48 is the more well known PS_GPIO6

I just received these pics from the "AISANRAY Radio Club Store" on Aliexpress where I bought mine.
Super helpful seller that probably asked the designer/producer

OZ1LAV - Steffen

[image]

[image]