plutoplus/plutoplus #55: pluto usb not showing on the my pc

A61BN, 2025-05-16:
pluto usb not showing on the my pc and not connecting to sdr console via usb or  ethernet i think it is a driver problem please help me solve this problem 

i have pluto plus  sdr

videobuff, 2025-05-17:
I assume you are using windows 10 or 11. 
Follow these steps and you will be ok. 

https://github.com/plutoplus/plutoplus/issues/51#issuecomment-2569582352

A61BN, 2025-05-17:
> I assume you are using windows 10 or 11. Follow these steps and you will be ok.
> 
> [#51 (comment)](https://github.com/plutoplus/plutoplus/issues/51#issuecomment-2569582352)

its working on other windows computer but still not connecting to sdr console but im geting the usb on the other windows machine
but on my main computer just a comport with no usb

A61BN, 2025-05-18:
> Weird. The W of windows. Could you find out which driver the p+ comport is using on the working computer. Look for the comport properties. You could even copy those driver files in my case serenum and serial dot sys to the other computer. They are in my case 32bit drivers. Met vriendelijke groetenErik SchottOp 17 mei 2025 om 12:31 heeft A61BN ***@***.***> het volgende geschreven:
> A61BN left a comment ([plutoplus/plutoplus#55](https://github.com/plutoplus/plutoplus/issues/55))
> 
> I assume you are using windows 10 or 11. Follow these steps and you will be ok.
> [#51](https://github.com/plutoplus/plutoplus/issues/51) (comment)
> 
> its working on other windows computer but still not connecting to sdr console but im geting the usb on the other windows machine
> but on my main computer just a comport with no usb

Still the same no usb and not connecting to sdr console

A61BN, 2025-05-25:
Still having the same problem but unable to connected via the network but I am facing a new problem the TX port is switching between tx1 and tx2 can you help me figure out the problem or maybe there is the setting to keep it to transmit on tx1