plutoplus/plutoplus #57: External Clock Source Instructions

smazokha, 2025-06-03:
Hey folks,

What are the instructions on how to use an external clock with a Pluto Plus?

I have a 10 MHz OCXO board plugged in using the U.FL connector, and the EXTCLK is connected to GND. However, when I turn on the board, it never boots (only green led is on). 

For reference, I am using [this OCXO board](https://www.amazon.com/Frequency-Oscillator-Temperature-Instrument-benchmarks/dp/B0BBVTPPHZ) and have original 0.32 firmware installed on the Pluto.

Thanks in advance!

SteffenLav, 2025-06-17:
Smazokha, did you ever get the external OCXO to work with Pluto Plus?

smazokha, 2025-07-07:
Hey @SteffenLav,

Apologies for the radio silence. Yes, I got it to work, but I used [this GPSDO](https://www.leobodnar.com/shop/index.php?main_page=product_info&cPath=107&products_id=393) instead to make sure that the other cheap board wasn't the issue. I managed to get the clock sync even with 10 MHz frequency, but haven't checked whether the phase sync will be consistent over time.

I relied on [this doc](https://wiki.amsat-dl.org/doku.php?id=en:tricks:pluto_extclk&do=export_pdf). See attached: [adalm_pluto_with_external_clock.pdf](https://github.com/user-attachments/files/21106626/adalm_pluto_with_external_clock.pdf)

## Here are the steps I took (I'm working with 3V3 board):

1. Turn off your Pluto+;
2. Short the EXTCLK with GND pins, connect the external clock via U.FL port on the board;
3. Enable external clock, configure to your desired frequency (I started wtih 25 MHz);
4. Enable Pluto Plus, SSH into it, and configure the device:

> `fw_setenv refclk_source external`
> `fw_setenv ad936x_ext_refclk_override "<25000000>"`
> `pluto_reboot reset`

5. After rebooting, your device should start using an external clock. 

## I verified that this setup worked as follows:

1. After these configurations, I set up simple IQ sample capture in GNUradio: Pluto SDR source -> QT GUI Waterfall
2. Configure your capture on a crowded WiFi channel
3. As you run your real-time FFT (i.e., waterfall), try connecting / disconnecting the external clock and you'll notice significant disruptions to your plot. If the external clock isn't configured correctly -- there won't be any disruptions (i.e., you're using internal clock and the U.FL port signal isn't picked up).

## To bring your device configs back to use the internal clock:

1. Disconnect the external clock and the EXTCLK pin short);
2. Modify the configurations:

> `fw_setenv refclk_source internal` 
> `fw_setenv ad936x_ext_refclk_override`
> `pluto_reboot reset`

3. The board will reboot, and you'll be back to normal mode afterwards.

SteffenLav, 2025-07-09:
Hey @smazokha,

Thank you so much for the thorough explanation - it all makes sense!